MEN-Mikro-Elektronik/13MD05-90 #13: MAIN_PR005222 FPGA_LOAD needs maintenance

michaelmiehling, 2018-07-23:
_From @michaelmiehling on July 23, 2018 5:52_

**Describe the bug**
FPGA_LOAD in MDIS should be updated. The current version 13Z100-91 is 1.11 with gpga_load.c 1.28 and there currently isnt a more up to date version. When dumping chameleon tables, there are often unidentifiable modules. The ??? are likely the Z135 IPCore. The below example was pulled from a G229:

```
CHAMELEONV2_UNIT:
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
--- ------ ------------------------ --- ---- --- --- --- --- ---------- ----------
  0 0x0018 16Z024_SRAM                0    0   1  13  63   0 0x00000000 0xfb808000
  1 0x0045 16Z069_RST                 0    0   0  12  63   0 0x00000200 0xfb808200
  2 0x007f 16Z127_GPIO                0    0   0   8  63   0 0x00000300 0xfb808300
  3 0x007e 16Z126_SERFLASH            0    0   1  11  63   0 0x00000400 0xfb808400
  4 0x0001 16Z001_SMB                 0    0   0   6   5   0 0x00000600 0xfb808600
  5 0x001d 16Z029_CAN                 0    0   1  18   6   0 0x00000700 0xfb808700
  6 0x0087 ?                          0    0   0  16   0   0 0x00001000 0xfb809000
  7 0x0087 ?                          0    1   0  16   1   0 0x00002000 0xfb80a000
  8 0x0087 ?                          0    2   0  16   2   0 0x00003000 0xfb80b000
  9 0x0087 ?                          0    3   0  16   3   0 0x00004000 0xfb80c000
 10 0x0087 ?                          0    4   0  16   4   0 0x00005000 0xfb80d000
 11 0x004c 16Z076_QSPI                0    0   0  13   7   2 0x00000000 0xfb810000
 12 0x002c 16Z044_DISP                1    0   3   1  63   0 0x00000500 0xfb808500
 13 0x0018 16Z024_SRAM                1    1   1  13  63   1 0x00000000 0xfb800000

```
**Expected behavior**
All supported FPGA IPs shall be indicated by the name and not ???

**Screenshots and logs**

**Revision of 13MD05-90 and related driver or module where the bug was detected**
13Z100-91 1.11

**Additional context**

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#25_

KonradJPE, 2018-08-16:
We have checked all the modules that are available in MDIS repository and we have added missing device Z127GPIO in chameleon table (chameleon.h). 
We have also found that Z135 IPCore (0x0087)  is present in this table. Could you please provide more information about version of chameleon.h which was used to create results described in this issue ? Unfortunately we do not have Z135 IPCore hardware on our side to test. 

Could you please provide complete list of devices that should be present in fpga_load tool? 
After checking, and adding missing devices, we can generate an updated version of executable fpga_load and commit it.

FlorianRauh, 2018-08-24:
the Problem is that the tool "fpga_load" doesn´t know all current available IP cores which could be in an FPGA. So you see all the "?" . This is not a question terms of a driver/module. The fpga_load tool itself needs an update.

try something like a -> fpga_load 4d45 1a88 9b 0 -t <- on a G229

KonradJPE, 2018-09-03:
fpga_load tool has been updated in commit 249680a (on branch jpe-dev). 
Now 0x0087 IPCore is successfully detected.

closed 2018-09-03 by KonradJPE