MEN-Mikro-Elektronik/13MD05-90 #17: MAIN_PR005082 13MD05-90 SC24 template buggy

michaelmiehling, 2018-07-23:
_From @michaelmiehling on June 28, 2018 9:42_

**Describe the bug**
compiling a MDIS project with a SC24.tpl for example for a BC50M results in an error
```
++++++++ Building descriptor system +++++++++++
processing /home/build/men-mdis/system.dsc
*** /home/build/men-mdis/system.dsc: line 207: illegal integer SCAN_SMBNR1
/opt/menlinux/BUILD/MDIS/TPL/desc.mak:54: recipe for target 'DESC/system.bin' failed
make[1]: *** [DESC/system.bin] Error 1
/opt/menlinux/BUILD/MDIS/TPL/rules.mak:415: recipe for target 'system' failed
make: *** [system] Error 2

```
Reason is that the current sc24.tpl has a string "SCAN_SMBNR1" where a integer would be expected.

**To Reproduce**

**Expected behavior**

**Screenshots and logs**

**Revision of 13MD05-90 and related driver or module where the bug was detected**
13MD05-90 1.20

**Additional context**
[PR5082.zip](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2144684/PR5082.zip)

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#19_

KonradJPE, 2018-08-09:
You have attached sc24.tpl file which contain correct values of SMB_BUSNBR. Should these values be use in every case?

FlorianRauh, 2018-08-09:
No. The SC24 is an AMD board. This AMD board has 2 SMBusses. Therefore the idea was to create smb2_x enries dynamically. But somehow it didn´t Count up, I guess.
[system.dsc.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2274628/system.dsc.txt)

mlassota, 2018-08-22:
I've committed the change that could be solution, although I cannot tell it works cause we have no hardware to test it on. Please check if it works and let us know.

KonradJPE, 2018-09-07:
Tested by Moore Blake, and is working now.

closed 2018-09-07 by KonradJPE