MEN-Mikro-Elektronik/13MD05-90 #21: MAIN_PR004581 MDIS for Linux must be updated

michaelmiehling, 2018-07-23:
_From @michaelmiehling on June 28, 2018 9:35_

**Describe the bug**
Detected issues:

1) Included 13Z017-06 MDIS5 driver for 16Z034_GPIO, 16Z037_GPIO,  16Z127_GPIOz17 is buggy (e.g. can not be used with F215, G215, ...). The driver is already fixed (git commit 62ebe4267ce from 13 Jul 2016) but not updated in 13md05-90.

2) Other MDIS drivers are outdated

3) Newer MDIS drivers missing (e.g. ARINC 13Z146/147)

4) User Manual update required:
    - A2: System Package contains all available MDIS drivers, not only the M43 M-Module driver
    - A2.1: Installation instruction erroneous: no mdis5linux folder after 13md0590.zip unzip
    - A 6 Installation of Kernel Sources outdated, the customer requires more detail info to succeed. It takes me about one day to get MDIS drivers for G215 with G25A running (with support from T. Schnürer and C. Zoz, see attached e-mail).
    - General manual review and update required!
--- Manual Errors reported from C. Zoz ---
- P10 A1.2.1
  1) optional tools: kdesu
     a) it is not optional if you want wdiswiz to scan the system
     b) in the end it uses gksudo
 
- P13 A2.1
  1) 13md0590.zip should be 13md05-90.zip
  2) The archive is flat and does not contain a folder (bad in general)
     ==> cd mdis5linux is wrong

- P14 A3.1
  1) "The previous examples": there are no previous examples
  2) Prerequisites should be handled in a previous section
  3) There is no pointer to /opt/menlinux/BIN/mdiswiz
  
- P17 A3.2 Creating a Manual MDIS P. (better Manually Creating a MDIS P.)
  It describes many settings that cannot be changed after creating a project via scanning
  - selfhosted (ok, not selfhostet makes little sense when scanning)
  - CPU platform

- nothing about how to build
- build requirements not complete: make scripts is missing as well (ok, there was a "do a make all" somewhere)
- how to install

5) MDIS Wizard erroneous and outdated
--- MDISWIZ Errors reported from C. Zoz ---
- initial popup: the warning in red is strange. And 'klick' is wrong.
  "To generate an initial config (scan system) klick Cancel and use button 'scan' ..."
- gksudo: command not found
  And in the next line it says 'finished' instead of 'error'
- LIN_KERNEL_DIR: " file autoconf.h not found in /include/linux or /include/generated. You must compile this kernel first. Come back later!
  I did no see a note somewhere about that
  Also you don't need a full compile: 'make prepare' seems sufficient
  *** on P19: there is a hint on 'make prepare'
- Project/Settings -> DEVNODE_INSTALL_DIR
  /dev is not persistent nowadays
- Project/Save_as --> Cancel pops up an error message

**To Reproduce**

**Expected behavior**

**Screenshots and logs**

**Revision of 13MD05-90 and related driver or module where the bug was detected**
 13MD05-90 01_19

**Additional context**
[PR4581.zip](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2144646/PR4581.zip)

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#15_

KonradJPE, 2018-10-15:
1.  13Z017-06 MDIS5 driver is updated on 13MD05-90 master branch - contains changes from commit 62ebe4267ce.

JSierl, 2019-02-05:
Done - the only open topic was moved to issue #73

closed 2019-02-05 by JSierl