MEN-Mikro-Elektronik/13MD05-90 #23: MAIN_PR004370 13md05-90_z001_crashes_immediately

michaelmiehling, 2018-07-23:
_From @michaelmiehling on June 28, 2018 9:31_

**Describe the bug**
z001 crashes at load.
sudo modprobe men_lx_z001_io

good: 13md05-90_1.19_beta8.zip
bad: 13md05-90_1.19_beta10.zip

```
dmesg:
 MEN SMB controller driver for IP core 16Z001_SMB
[  479.027368] BUG: unable to handle kernel NULL pointer dereference at 0000000000000028
[  479.027479] IP: [<ffffffffc0008080>] z001_smb_init+0x80/0x1000 [men_lx_z001_io]
[  479.027573] PGD 0
[  479.027607] Oops: 0000 [#1] SMP
[  479.027659] Modules linked in: men_lx_z001_io(OE+) nfsv3 nfs_acl rpcsec_gss_krb5 auth_rpcgss nfsv4 nfs lockd grace fscache xt_CHECKSUM iptable_mangle ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp bridge stp llc ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter ip_tables x_tables binfmt_misc zfs(PO) zunicode(PO) zcommon(PO) znvpair(PO) spl(O) zavl(PO) intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp crct10dif_pclmul crc32_pclmul ghash_clmulni_intel cryptd acpi_als mei_me shpchp lpc_ich mei kfifo_buf industrialio mac_hid kvm_intel kvm irqbypass sunrpc autofs4 uas usb_storage amdkfd amd_iommu_v2 radeon igb ttm drm_kms_helper syscopyarea sysfillrect sysimgblt
[  479.028844]  dca e1000e fb_sys_fops ahci ptp libahci drm pps_core i2c_algo_bit video fjes [last unloaded: men_lx_z001_io]
[  479.029028] CPU: 1 PID: 9738 Comm: modprobe Tainted: P           OE   4.4.0-87-generic #110-Ubuntu
[  479.029129] Hardware name: MEN GmbH MEN G22/F22P/CB70/A22/MEN G22/F22P/A22/CB70, BIOS G22 1.23 08/29/2014
[  479.029235] task: ffff88003594b400 ti: ffff88007734c000 task.ti: ffff88007734c000
[  479.029319] RIP: 0010:[<ffffffffc0008080>]  [<ffffffffc0008080>] z001_smb_init+0x80/0x1000 [men_lx_z001_io]
[  479.029437] RSP: 0018:ffff88007734fc80  EFLAGS: 00010286
[  479.029499] RAX: 0000000000000000 RBX: ffff8800773f0400 RCX: 0000000000000000
[  479.029579] RDX: 0000000000006a18 RSI: 00000000024000c0 RDI: ffff8800773f0744
[  479.029660] RBP: ffff88007734fc88 R08: ffff88007ce5a140 R09: ffff88007a403500
[  479.029740] R10: ffff8800773f0400 R11: 0000000000000312 R12: ffff8800773badc0
[  479.029820] R13: 0000000000000000 R14: ffffffffc0008000 R15: ffff880076c0f7e0
[  479.029901] FS:  00007fdfe7687700(0000) GS:ffff88007ce40000(0000) knlGS:0000000000000000
[  479.029992] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  479.030058] CR2: 0000000000000028 CR3: 0000000078e99000 CR4: 00000000001406e0
[  479.030138] Stack:
[  479.030165]  ffffffff81e11080 ffff88007734fd08 ffffffff81002123 ffffffff8183dfa8
[  479.030271]  00000000024000c0 ffff88007734fcb8 ffffffff8183dfdc ffff88007734fd08
[  479.030375]  ffffffff811ed6e3 0000000000000001 ffffffffc0966040 0000000000000018
[  479.030480] Call Trace:
[  479.030520]  [<ffffffff81002123>] do_one_initcall+0xb3/0x200
[  479.030590]  [<ffffffff8183dfa8>] ? preempt_schedule_common+0x18/0x30
[  479.030665]  [<ffffffff8183dfdc>] ? _cond_resched+0x1c/0x30
[  479.030734]  [<ffffffff811ed6e3>] ? kmem_cache_alloc_trace+0x183/0x1f0
[  479.030816]  [<ffffffff8118df83>] do_init_module+0x5f/0x1cf
[  479.030885]  [<ffffffff8110ab0f>] load_module+0x166f/0x1c10
[  479.030951]  [<ffffffff811070b0>] ? __symbol_put+0x60/0x60
[  479.031020]  [<ffffffff81215360>] ? kernel_read+0x50/0x80
[  479.031086]  [<ffffffff8110b2f4>] SYSC_finit_module+0xb4/0xe0
[  479.031155]  [<ffffffff8110b33e>] SyS_finit_module+0xe/0x10
[  479.031223]  [<ffffffff81841eb2>] entry_SYSCALL_64_fastpath+0x16/0x71
[  479.031297] Code: 51 96 c0 e8 7a 5a 18 c1 e9 f0 00 00 00 48 c7 43 20 40 60 96 c0 c7 43 28 01 00 00 00 48 8d bb 44 03 00 00 48 c7 43 30 00 60 96 c0 <48> 8b 04 25 28 00 00 00 48 c7 c6 fa 51 96 c0 8b 0d 83 df 95 00
[  479.031862] RIP  [<ffffffffc0008080>] z001_smb_init+0x80/0x1000 [men_lx_z001_io]
[  479.031954]  RSP <ffff88007734fc80>
[  479.031995] CR2: 0000000000000028
[  479.032086] ---[ end trace a6090b26f5494530 ]---

```
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

**Screenshots and logs**

**Revision of 13MD05-90 and related driver or module where the bug was detected**
 13MD05-90 1.20-rc2

**Additional context**

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#13_

KonradJPE, 2018-08-09:
Could you please write where we can find sources:
good: 13md05-90_1.19_beta8.zip
bad: 13md05-90_1.19_beta10.zip ?

KonradJPE, 2018-08-22:
Solved. 13Z001-90 submodule in commit id f48be07ef6aed1236e355f0f1c4319dec5be01d8

closed 2018-08-22 by KonradJPE