MEN-Mikro-Elektronik/13MD05-90 #28: MAIN_PR004262 13MD05-90 mdiswiz system_scan.sh

michaelmiehling, 2018-07-23:
_From @michaelmiehling on June 28, 2018 9:10_

**Describe the bug**
Execution of skript scan_system.sh behaves different to scan via mdiswiz gui.
Differnt system.dsc files are generated (see attachement).
Additionally slot nr of G215 which is in slot5 is after mdiszwiz executed scan detected in slot2. Result is, that G204 can not be configured to slot2 because already occupied.

- Slot 1: F23P
- Slot 2: G204 + M36N
- Slot 3: G204 + M58
- Slot 4: G204 + M62N
- Slot 5: G215 

F204 and G204 seem to be not involved if scan_system.sh is executed.

**To Reproduce**

**Expected behavior**

**Screenshots and logs**
[PR4262.zip](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2144549/PR4262.zip)

**Revision of 13MD05-90 and related driver or module where the bug was detected**
 13MD05-90 1.20rc2

**Additional context**

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#8_

KonradJPE, 2018-08-09:
Is it possible to provide more information how to reproduce behaviour described in this issue? 
For our knowledge MDIS wizard is using exactly the same scan_system.sh script during ./INSTALL. File is simply copied into /opt/menlinux/ and then used by Wizard. 
After checking on our machine we are getting exactly the same system.dsc files.

FlorianRauh, 2018-08-09:
hmm, Im not sure if we understand each other.
There are two way of scanning the System
a) via console /opt/menlinux/scan_system.sh
b) using the gui and clicking the button.
[image]
I got different results of the System.dsc if did a) or b)

KonradJPE, 2018-09-10:
Detection of boards F205 and G204 have been added into scan_system script. Updated mdiswiz executable that is required to open updated system.dsc file is in commit id: b296745

closed 2018-09-10 by KonradJPE