MEN-Mikro-Elektronik/13MD05-90 #37: I2C does not compile on BL70

michaelmiehling, 2018-07-23:
_From @michaelmiehling on July 4, 2018 9:10_

**Describe the bug**
With MDIS project on BL70 including I2C drivers, the build generates error message and stops (first log)
If a recompile was performed, it worked without error.
After removal of the I2C drivers in the mdiswiz, another error appears (second log)

**To Reproduce**

**Expected behavior**
I2C driver of Intel CPUs should compile without error

**Screenshots and logs**
first log:
```
Directory OBJ/dbg/men_bb_smb2 created
/opt/menlinux/BUILD/MDIS/TPL/component.mak:50: /opt/menlinux/DRIVERS/I2C_BUSSES/i2cpiix4/driver.mak: No such file or directory
make[2]: *** No rule to make target '/opt/menlinux/DRIVERS/I2C_BUSSES/i2cpiix4/driver.mak'.  Stop.
/opt/menlinux/BUILD/MDIS/TPL/rules.mak:544: recipe for target 'DRIVERS/I2C_BUSSES/i2cpiix4/driver.mak' failed
make[1]: *** [DRIVERS/I2C_BUSSES/i2cpiix4/driver.mak] Error 2
/opt/menlinux/BUILD/MDIS/TPL/rules.mak:470: recipe for target 'dbg' failed
make: *** [dbg] Error 2

```

second log:
```
/home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c: In function ‘men_OSS_SemWait’:
/home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c:241:3: error: unknown type name ‘wait_queue_t’
   wait_queue_t __wait;
   ^
/home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c:245:24: error: passing argument 1 of ‘init_waitqueue_entry’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   init_waitqueue_entry(&__wait, current); 
                        ^
In file included from ./include/linux/mmzone.h:9:0,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/kmod.h:22,
                 from ./include/linux/module.h:13,
                 from /opt/menlinux/LIBSRC/OSS/oss_intern.h:90,
                 from /home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c:53:
./include/linux/wait.h:77:20: note: expected ‘struct wait_queue_entry *’ but argument is of type ‘int *’
 static inline void init_waitqueue_entry(struct wait_queue_entry *wq_entry, stru
                    ^
/home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c:247:29: error: passing argument 2 of ‘add_wait_queue’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   add_wait_queue( &sem->wq, &__wait);
                             ^
In file included from ./include/linux/mmzone.h:9:0,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/kmod.h:22,
                 from ./include/linux/module.h:13,
                 from /opt/menlinux/LIBSRC/OSS/oss_intern.h:90,
                 from /home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c:53:
./include/linux/wait.h:148:13: note: expected ‘struct wait_queue_entry *’ but argument is of type ‘int *’
 extern void add_wait_queue(struct wait_queue_head *wq_head, struct wait_queue_e
             ^
/home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c:291:32: error: passing argument 2 of ‘remove_wait_queue’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   remove_wait_queue( &sem->wq, &__wait);
                                ^
In file included from ./include/linux/mmzone.h:9:0,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/kmod.h:22,
                 from ./include/linux/module.h:13,
                 from /opt/menlinux/LIBSRC/OSS/oss_intern.h:90,
                 from /home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.c:53:
./include/linux/wait.h:150:13: note: expected ‘struct wait_queue_entry *’ but argument is of type ‘int *’
 extern void remove_wait_queue(struct wait_queue_head *wq_head, struct wait_queu
             ^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.o' failed
make[4]: *** [/home/men/13MD05-90/OBJ/dbg/men_oss/oss_sem.o] Error 1
scripts/Makefile.build:581: recipe for target '/home/men/13MD05-90/OBJ/dbg/men_oss' failed
make[3]: *** [/home/men/13MD05-90/OBJ/dbg/men_oss] Error 2
Makefile:1550: recipe for target '_module_/home/men/13MD05-90/OBJ' failed
make[2]: *** [_module_/home/men/13MD05-90/OBJ] Error 2
/opt/menlinux/BUILD/MDIS/TPL/rules.mak:568: recipe for target 'callkernelbuild' failed
make[1]: *** [callkernelbuild] Error 2
/opt/menlinux/BUILD/MDIS/TPL/rules.mak:470: recipe for target 'dbg' failed
make: *** [dbg] Error 2

```

**Revision of 13MD05-90 and related driver or module where the bug was detected**
 13MD05-90 05_01_22 with kernel 4.13 of Ubuntu 16.04.4

**Additional context**

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#22_

michaelmiehling, 2018-07-23:
This issue was moved to MEN-Mikro-Elektronik/13MD05-90#16

mlassota, 2018-09-21:
Second log problem can be related or even fixed in 0abb43c commit.

KonradJPE, 2018-10-10:
Compiling without problem on Ubuntu 16.04 kernel v4.15, 18.04 kernel v4.15, CentOS 7.5

closed 2018-10-10 by KonradJPE