MEN-Mikro-Elektronik/13MD05-90 #38: tools are missing in make file if generated with mdiswiz for BL70

michaelmiehling, 2018-07-23:
_From @michaelmiehling on July 4, 2018 9:44_

**Describe the bug**
If for BL70 a mdiswiz configuration is done per scan, the tools are not per default added to the makefile: I2C, MSCAN, z17 ...
If drivers are added manually, the tools are added to the makefile as expected

**To Reproduce**
workaround is to add the tools manually and recompile the project

**Expected behavior**
per default, all tools related to the configured drivers of the complete system shall be activated for make

**Screenshots and logs**

**Revision of 13MD05-90 and related driver or module where the bug was detected**
13MD05-90 05_01_21 kernel 4.10.0-42

**Additional context**

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#23_

michaelmiehling, 2018-07-23:
This issue was moved to MEN-Mikro-Elektronik/13MD05-90#15

KonradJPE, 2018-10-23:
Duplicate of issue #15.

closed 2018-10-23 by KonradJPE