MEN-Mikro-Elektronik/13MD05-90 #39: MAIN_PR003403 MDIS 1.18 with CentOS 6.8 x64 doesn't load modules automatically

michaelmiehling, 2018-07-23:
_From @michaelmiehling on July 18, 2018 5:54_

**Describe the bug**
When you want to load automatically the drivers as mentionned in the MDIS users' manual §A 4.3, the modules are not loaded automatically when accessing the /dev/mdis device node for the first time.

When you use MDIS 13MD05-90 1.17, the modules are correctly loaded when accessing the /dev/mdis device node for the first time.

rules.mak 1.17 :

```
installdevnode:
ifeq ($(HASDEVFS),1)
else
 ifdef DEVNODE_INSTALL_DIR
  ifdef MDIS_MAJOR_NUMBER
	$(Q)install --directory $(DEVNODE_INSTALL_DIR)
	$(Q)rm -f $(DEVNODE_INSTALL_DIR)/mdis
	$(MKNOD) $(DEVNODE_INSTALL_DIR)/mdis c $(MDIS_MAJOR_NUMBER) 0
	$(Q)chmod 666 $(DEVNODE_INSTALL_DIR)/mdis
	$(Q)if test -d /lib/udev/devices; then cp -a $(DEVNODE_INSTALL_DIR)/mdis /lib/udev/devices/; fi
  endif
 endif
endif

was modified to this in 1.18 :

installdevnode:

ifeq ($(HASDEVFS),1)
else
 ifdef DEVNODE_INSTALL_DIR
	$(Q)install --directory $(DEVNODE_INSTALL_DIR)
	@$(ECHO) "============================================================================"
	@$(ECHO) "=> When rebuilding modified modules rmmod old modules before new modprobe <="
	@$(ECHO) "============================================================================"
 endif
endif

```
If you have not copied the /dev/mdis created with mknode, this prevent to load automatically the driver as mentioned in the users' manual.

This probably affect every Linux distributions.

**To Reproduce**

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

**Revision of 13MD05-90 and related driver or module where the bug was detected**
 13MD05-90 01_18

**Additional context**
Workaround - Manually create a /dev/mdis devnode with mknode and copy it to /lib/udev/devices :

```
# mknode /dev/mdis c MDIS_MAJOR_NB 0
# chmod 666 /dev/mdis
# cp -a /dev/mdis /lib/udev/devices

```
where MDIS_MAJOR_NB is the one entred in MDIS wizard (e.g: 248 ).

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#24_

michaelmiehling, 2018-07-23:
This issue was moved to MEN-Mikro-Elektronik/13MD05-90#14

mlassota, 2018-09-21:
This seems to be issue #24 duplicate.

closed 2018-09-21 by mlassota