MEN-Mikro-Elektronik/13MD05-90 #40: MAIN_PR005083 13MD05-90 compiling under Debian

michaelmiehling, 2018-07-23:
_From @michaelmiehling on June 28, 2018 9:43_

**Describe the bug**
Currently MDIS5 is not prepared to be hosted on a Debian Plattform.
Debian uses 2 header directories
/usr/src/linux-4.9.0.x-amd64	
/usr/src/linux-4.9.0.x-common

MDIS is not able to find the right .h files in both folders during compiling a project.

**To Reproduce**

**Expected behavior**

**Screenshots and logs**

**Revision of 13MD05-90 and related driver or module where the bug was detected**
 13MD05-90 1.20

**Additional context**

_Copied from original issue: MEN-Mikro-Elektronik/13MD05-90_old#20_

mlassota, 2018-08-03:
Debian's approach doesn't seem to be any different from that in Ubuntu, though it is. On both Debian and Ubuntu there are two directories for kernel headers - one for architecture dependent part, and other for architecture independend part. On Debian architecture independent headers directory is postfixed with '-common' and on Ubuntu with '-generic'. Architecture dependent headers directory is postfixed with architecture name (e.g. '-amd64') on Debian, but on Ubuntu ther is no postfix whatsoever. Nevertheless to get proper headers for MDIS on Debian and Ubuntu one should make symlink '/usr/src/linux' to point to '/usr/src/linux-headers-$(uname -r)'. Therefore on Debian symlink will point to architecture dependent headers dir, while on Ubuntu it will point to architecture independent headers dir. Having this in mind one should have no problem to compile MDIS on Debian and Ubuntu.

MDIS needs to have some kernel flags in order to compile, but on Debian the makefile where they can be found is in different directory e.i. '/usr/src/linux-headers-$(uname -r)/source', therefore a distribution detection has been added to MIDS makefile so this directory could be handled properly. However Debian's kernel headers seem to have problem on they own which results in some warnings written on output because of some missing makefiles, so MDIS's makefiles refactorization was made and this messy output redirected to /dev/null.

KonradJPE, 2019-03-18:
MDIS functional tests have been performed on debian 4.9.0-8-amd64.

closed 2019-03-18 by KonradJPE