MEN-Mikro-Elektronik/13MD05-90 #44: mdiswizard formatting error / font not supported

HIttner, 2018-09-14:
I tested the latest commit in Lubuntu 18.04 and the mdis wizard opens with messed up text lines (see screenshot for kernel version and picture of behavior)

[image]

mlassota, 2018-09-28:
Fixed

closed 2018-09-28 by mlassota

mlassota, 2018-10-10:
It turned out that pkexec is not a good replacement for gksudo as it does change user directory to root directory if not configured. And it's hard to expect from user to have it configured earlier as it requires to create configuration xml file for each application that needs privilege escalation - this is pkexec policy.
Therefore usage of gksudo and pkexec has been removed.
If a user wants to use scanner in MDIS Wizard he needs to run wizard with 'sudo', otherwise there will be a proper message in scanner that it needs to be run with said 'sudo'.