MEN-Mikro-Elektronik/13MD05-90 #47: Error when saving/building projects as a normal user

bmoore84, 2018-09-19:
**Description**
Error generated when saving or building project as normal user.
Despite the error, it seems the configurations are still saved/built.
When running wizard as root, this issue does not occur. 

**Screenshots**
[image]

**Expected behavior**
Save/build project as a normal user without error

**Where detected**
Ubuntu 16.04.5 Kernel 4.15

mlassota, 2018-11-09:
There have been couple of changes in MDIS Wizard since you've reported this error, thus it might have been solved before I looked at it, because I was unable to reproduce it.
Could you please try to recreate it with the latest binary?

bmoore84, 2018-11-09:
Hello Marcin,
I will test again and see what is different.
Thanks for the update.

Best regards,

Blake

Blake Moore
Customer Support
MEN Mikro Elektronik GmbH
Neuwieder Straße 3-7
90411 Nürnberg, Germany
Phone +49 911 99 33 5 - 253
Fax +49 911 99 33 5 - 901
Blake.Moore@men.de<mailto:Blake.Moore@men.de>
www.men.de<http://www.men.de>

> Subscribe to our newsletter<https://www.men.de/news-media/newsletter/>

MEN Mikro Elektronik GmbH - Bernd Härtlein (CEO) - Yilmaz Kocak (CFO)  - Handelsregister/Trade Register AG Nürnberg HRB 5540
Please consider the environment before printing this e-mail

closed 2019-02-06 by bmoore84