MEN-Mikro-Elektronik/13MD05-90 #49: Unable to start mdiswiz in Lubuntu 16.04.03

bmoore84, 2018-10-11:
**Description**
When attempting to start mdiswiz, we are met with errors that prevent the wizard from starting

**Example of Issue**
men@men-VirtualBox:~/work/A21_2/mdis_A21B_M77$ sudo /opt/menlinux/BIN/mdiswiz
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
mdiswiz: ../../src/xcb_conn.c:195: write_vec: Assertion `!c->out.queue_len' failed.
Aborted (core dumped)

men@men-VirtualBox:~/work/A21_2/mdis_A21B_M77$ uname -r
4.4.0-36-generic

**Expected behavior**
Executing command to start wizard should not cause any errors and allow the wizard to load

**Where detected**
Lubuntu 16.04.03 Kernel 4.4.0-36-generic (Flo's VirtualBox Environment)

mlassota, 2018-10-11:
Unfortunately I was unable to reproduce this error, and I'm talking here about mdiswiz being aborted and core being dumped.

As for the fontconfig errors, well, mdiswiz is statically linked with an old fonconfig library (fontconfig-2.8.0@2010-12-16) and you run it on a new OS (lubuntu-16.04.3-desktop-amd64@2017-08-01). My point is that configuration files for fontconfig are not 100% compatible with such old fontconfig. I could try to link mdiswiz with newer fontconfig but I can't guarantee it will run on old distros, which is one of the requirements for mdiswiz. It's up to you to decide.

Back to core dump, could you please provide more info about: the commit the mdiswiz was taken from, Makefile and system.dsc, os architecture (is it 32- or 64-bit). Because I downloaded lubuntu-16.04.3-desktop-amd64.iso, installed on virtual box and the first thing after I booted it was to download mdiswiz (commit 09de7b5311507d503afef6f8e5244edfdf1d6563) and run it and it worked like a charm.  Then I downloaded kernel version you have, rebooted, run it and again it worked.

bmoore84, 2018-10-12:
After further testing, Flo and I agree it is ok to close this issue. Upon installing a fresh Lubuntu 16.04, and installing MDIS, the wizard does load. Our VMs are upgrades from 14.04 and a lot of changes were made within the VMs.

closed 2018-10-12 by bmoore84