MEN-Mikro-Elektronik/13MD05-90 #53: Improve install script usability

JSierl, 2018-10-12:
to be defined in detail by Florian Rauh

FlorianRauh, 2018-10-12:
1.
During ./INSTALL a message appears
"copy .git to /opt/menlinux...ok"
why is this being done? Does it make sense.

2. Why am I being ask to overwrite? 
"Overwrite folder /opt/menlinux/TOOLS/M_TMR (y/n/q): y"
Does this make sense?

mlassota, 2018-10-15:
> During ./INSTALL a message appears
> "copy .git to /opt/menlinux...ok"
> why is this being done? Does it make sense.
It doesn't. Having .git directories in installation point serves no purpose.
Fixed.

>     1. Why am I being ask to overwrite?
>        "Overwrite folder /opt/menlinux/TOOLS/M_TMR (y/n/q): y"
>        Does this make sense?
It doesn't.
Fixed.

closed 2018-10-15 by mlassota