MEN-Mikro-Elektronik/13MD05-90 #56: Unable to modprobe M77 module

bmoore84, 2018-10-19:
**Description**
System Setup:
G23 + G204 w/M77N

While performing steps:

for i in {0..7}; do mknod /dev/ttyD$i c 19 $i; done
modprobe men_bb_d203_a24
modprobe men_mdis_kernel
mdis_createdev -b d203_a24_1
modprobe men_lx_m77 devName=m77_1 brdName=d203_a24_1 slotNo=0 mode=2,2,2,2 echo=0,0,0,0

"modprobe men_lx_m77 devName=m77_1 brdName=d203_a24_1 slotNo=0 mode=2,2,2,2 echo=0,0,0,0" results in error:
[image]

Dmesg:
[image]

System.dsc file attached
[system.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2495764/system.txt)

lspci dump
[lspci.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2495780/lspci.txt)

**Expected behavior**
Create d203 device and add men_lx_m77 module successfully

**Where detected**
MDIS - Latest git pull from jpe-dev branch

Ubuntu 16.04.4 LTS
Kernel 4.15.0-041500-generic

CentOS 7.5.1804
3.10.0-862.14.4.el7.x86_64

mlassota, 2018-10-23:
Could you please pull latest changes from jpe-dev branch and check if the problem still occurs?
Maybe this is related to issue #57, so let system scan script create configuration.

bmoore84, 2018-10-23:
Unfortunately, its the same behavior. I have added a reply to issue#57 with system.dsc.

Thanks

Blake

Blake Moore
Customer Support
MEN Mikro Elektronik GmbH
Neuwieder Straße 3-7
90411 Nürnberg, Germany
Phone +49 911 99 33 5 - 253
Fax +49 911 99 33 5 - 901
Blake.Moore@men.de<mailto:Blake.Moore@men.de>
www.men.de<http://www.men.de>

> Subscribe to our newsletter<https://www.men.de/news-media/newsletter/>

MEN Mikro Elektronik GmbH - Bernd Härtlein (CEO) - Yilmaz Kocak (CFO)  - Handelsregister/Trade Register AG Nürnberg HRB 5540
Please consider the environment before printing this e-mail

mlassota, 2018-11-28:
It seems that the main cause the system scan script was unable to detect M77 was that G204 had its memory regions access disabled. Fresh OS install helped.

closed 2018-11-28 by mlassota