MEN-Mikro-Elektronik/13MD05-90 #57: System scan script does not detect M77N

bmoore84, 2018-10-19:
**Description**
Scan system script does not recognize M77N.
It will detect the G204 and add it as d203_a24_1, however you have to then start mdiswiz, load the project, and add the m77 module manually. 

System.dsc 
[system.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2496077/system.txt)

**Expected Behavior**
Scan system script should detect M77N and add module to system.dsc

**Where Detected**
CentOS Linux release 7.5.1804 Kernel 3.10.0-862.14.4.el7.x86_64

mlassota, 2018-10-23:
Fixed.

bmoore84, 2018-10-23:
Unfortunately, its the same behavior.
I removed MDIS and did a new git pull from jpe-dev and installed MDIS again. 
The carrier board is again detected but the script fails to detect the M77N
[system (2).txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2506367/system.2.txt)

mlassota, 2018-10-23:
Could you please run
sudo bash -vx /opt/menlinux/scan_system.sh /opt/menlinux > scan.log 2>&1
and send me 'scan.log'?

I've tested it with CentOS 7.5 3.10.0-862.11.6.el7.x86_64 on F23P+G204+M77N and it worked fine.

bmoore84, 2018-10-23:
I have attached the log. 
[scan.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2506479/scan.log)

Thank you

mlassota, 2018-11-28:
It seems that the main cause the system scan script was unable to detect M77 was that G204 had its memory regions access disabled. Fresh OS install helped.

closed 2018-11-28 by mlassota