MEN-Mikro-Elektronik/13MD05-90 #60: men_lx_z77 cannot be compiled in linux kernel > 4.8

HIttner, 2018-10-26:
Using the current jpe-dev branch, it is not possible to build an mdis project containing a mezzcham device including at least one z87 ethernet IP core in Ubuntu 18.04 kernel 4.15.

A colleague from MEN already patched this driver ( can be found in 
`13MD05-90-dev\13MD05-90-dev\DRIVERS\ETH_16Z077\DRIVER`)
Please include in jpe-dev branch and verify correct functionality of both interfaces.

Hardware is included in MDIS Test system 1 (sidecard of F26L F614).
The FPGA on F614 contains two z87 IP cores which provide fast ethernet at the front (D-Sub).
Adapter cables are included to connect to RJ45

mlassota, 2018-10-31:
I've merged this patched driver to jpe-dev branch but right after men_lx_z77 driver is loaded the computer freezes (F23P+Ubuntu 18.04.1+kernel 4.15.0).
On F23P+Centos 7.5.1804+kernel 3.10.0 the driver does not even compile.
On F23P+Ubuntu 14.04.4+kernel 4.2.0 the driver compiles and the computer does not freez after driver being loaded.

HIttner, 2018-10-31:
sorry, some internal misunderstanding.
https://github.com/MEN-Mikro-Elektronik/13Z077-90/tree/development 
Please try this one

mlassota, 2018-11-06:
I've merged and tested this driver on:
F23P+F614+Ubuntu 18.04.1 64-bit+kernel 4.15.0
F23P+F614+Centos 7.5.1804 64-bit+kernel 3.10.0
F23P+F614+Ubuntu 16.04.5 64-bit+kernel 4.15.0
F23P+F614+Ubuntu 14.04.4 64-bit+kernel 4.2.0
and it compiles without errors and Internet connection also works fine in every tested case.

closed 2018-11-06 by mlassota