MEN-Mikro-Elektronik/13MD05-90 #61: History '*.txt' files for main repository and module repositories - do we need them?

mlassota, 2018-11-09:
Currently the installation script creates history files for main repository and for module repositories. To do that it requires to have the '.git' directory which is a git repository data, but when the MDIS sources come from archive file (e.g. 13MD05-90_01_21.zip) the '.git' directory is not included (for obvious reasons), therefore installation script fails.

I propose to remove this part of code responsible for creating history files, unless they are really needed.

dpfeuffer, 2018-11-28:
The git info extracted from INSTALL and saved in *_history/url/version.txt shall still be available to provide module information .
The git tag info shall also be added (*_tag.txt) and the rules.mak modified to use the *_tag.txt file to set the GIT_VERSION during make (instead of using git describe). Currently, make requires to have .git under /opt/menlinux because of the use of git in rules.mak. We have to omit .git in /opt/menlinux. 

Suggestion to achieve this:
1) If .git folder is available in <13MD05-90 repo dir>: INSTALL calls <make-history-script> to extract git information and create *_history/url/version/tag.txt under <13MD05-90 repo dir>/HISTORY.
2) INSTALL copies <13MD05-90 repo dir>/HISTORY to $MENLINUX_ROOT
3) Before creating a release package (e.g. 13MD05-90_01_21.zip):
   a) the <make-history-script> must be called to create the HISTORY
   b) the .git folder must be deleted manually or from a <make-delivery-script> but not from <make-history-script>.
4) <make-delivery-script> could be part of 13MD05-90:
  a) copy the entire 13MD05-90 content to a temp folder
  b) call <make-history-script> to create HISTORY inside the temp folder
  c) remove <make-delivery-script> from the temp folder
  d) create archive from the temp folder

KonradJPE, 2019-03-14:
Fixed.

closed 2019-03-14 by KonradJPE