MEN-Mikro-Elektronik/13MD05-90 #62: Makefile/system.dsc generated from scan_system.sh vs generated from mdiswiz

dpfeuffer, 2018-11-16:
scan_system.sh/mdiswiz @13MD05-90_01_21:
The Makefile/system.dsc files generated from scan_system.sh are quite different as the files saved from mdiswiz:
1) a lot of unnecessary spacing/newline differences

2) system.dsc: mdiswiz converts some (but not all) DEVICE_IDV2_X entries into DEVICE_ID_X entries. Why?

3) Makefile generated from scan_system.sh: Wrong header:    
# Generated by mdiswiz 2.01.022-linux-10.3
# 2012-12-20

4) Makefile: very different SW modules will be built. E.g.
scan_system.sh generated:
ALL_LL_DRIVERS = \
     XM01BC/DRIVER/COM/driver.mak\
     SMB2/DRIVER/COM/driver.mak\
     Z017/DRIVER/COM/driver.mak\
     Z017/DRIVER/COM/driver_io.mak\
     Z017/DRIVER/COM/driver_z127.mak\
     Z017/DRIVER/COM/driver_z127_noirq.mak\
     Z017/DRIVER/COM/driver_z127v01.mak\
     MSCAN/DRIVER/COM/driver_z15.mak\
     MSCAN/DRIVER/COM/driver_z15_io.mak\
     MSCAN/DRIVER/COM/driver_odin.mak\
#SCAN_NEXT_LL_DRIVER
mdiswiz generated:
ALL_LL_DRIVERS = \
	SMB2/DRIVER/COM/driver.mak \
	XM01BC/DRIVER/COM/driver.mak \
	Z017/DRIVER/COM/driver.mak \
	MSCAN/DRIVER/COM/driver_z15.mak

You can reproduce this with:
1) start mdiswiz and generate the system configuration (Makefile/system.dsc) with the mdiswiz scan feature (that calls scan_system.sh), then close the mdiswiz
2) start mdiswiz, save the configuration without any modifications, then close the mdiswiz
3) compare Makefile with Makefile.bak and system.dsc with system.dsc.bak

How to explain a customer that a simple open/save/close of an MDIS project will result into different MDIS Makefile/system.dsc and build SW modules?
The scan functionality should be completely revised (usage, implementation).

JSierl, 2019-02-05:
Isn't this solved already? If so, please set it done.

KonradJPE, 2019-04-03:
> 3. Makefile generated from scan_system.sh: Wrong header:

What should the correct header look like? 
What about: 
**Generated by scan_system.sh script version: \<commitId\>** ?

dpfeuffer, 2019-04-05:
"Generated by scan_system.sh script version: <commitId>" looks good.
But how to specify <commitId>? Do you want to modify the scan_system.sh header during INSTALL?

KonradJPE, 2019-04-26:
> Generated by scan_system.sh script version: \<commitId\> looks good.
> But how to specify \<commitId\>? Do you want to modify the scan_system.sh header during INSTALL?

\<commitId\> will be taken from HISTORY that is copied/created together with MDIS sources while ./INSTALL script is invoked. 
While running scan_system.sh,  line: 
**Generated by scan_system.sh script version: <commitId>** 
will be added at the beginning of the Makefile/system.dsc file.

KonradJPE, 2019-04-29:
Header like below will be added into Makefile, system.dsc file by scan_system.sh script: 

> Generated by scan_system.sh script version: 
> 71b858c79522d64f8468fe2ec7dec53700b12d34
> Mon Apr 29 11:57:14 CEST 2019

KonradJPE, 2019-04-29:
For clarification, I have divided this issue: 

> 1. a lot of unnecessary spacing/newline differences

Fixed

> 2. system.dsc: mdiswiz converts some (but not all) DEVICE_IDV2_X entries into DEVICE_ID_X entries. Why?

Moved to issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/107

> 3. Makefile generated from scan_system.sh: Wrong header:

Fixed

> 4. Makefile: very different SW modules will be built. E.g.
> scan_system.sh generated: ...

This is due differences in xml parser in mdiswiz, and xml parser in scan_system.sh script. 
See issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/85

closed 2019-04-29 by KonradJPE