MEN-Mikro-Elektronik/13MD05-90 #65: sysfs: cannot create duplicate filename

AndreasGeissler, 2018-12-18:
When using a system with two or more of the same FPGA programming file MDIS will crash with:

```
[    5.371765] sysfs: cannot create duplicate filename '/devices/men_chameleon/CC10-02IC200'
[    5.371775] WARNING: CPU: 5 PID: 303 at /build/linux-CuEUJv/linux-4.15.0/fs/sysfs/dir.c:31 sysfs_warn_dup+0x56/0x70
[    5.371776] Modules linked in: men_lx_chameleon(OE+) men_chameleon_io(OE) men_chameleon(OE) men_oss(OE) men_dbg(OE) mii parport_pc ppdev lp parport ip_tables x_tables autofs4 uas usb_storage i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops e1000e drm ahci libahci i2c_i801 ptp pps_core video
[    5.371793] CPU: 5 PID: 303 Comm: systemd-modules Tainted: G           OE    4.15.0-38-generic #41-Ubuntu
[    5.371794] Hardware name: MEN eHuronRiver/MEN F21P, BIOS F21P 2.11 11/01/2016
[    5.371796] RIP: 0010:sysfs_warn_dup+0x56/0x70
[    5.371797] RSP: 0018:ffffa87f80a238b8 EFLAGS: 00010286
[    5.371799] RAX: 0000000000000000 RBX: ffff9b5fbc21c000 RCX: 0000000000000006
[    5.371799] RDX: 0000000000000007 RSI: 0000000000000096 RDI: ffff9b5fce356490
[    5.371800] RBP: ffffa87f80a238d0 R08: 0000000000000001 R09: 00000000000002e6
[    5.371801] R10: 0000000000000001 R11: 0000000000000000 R12: ffff9b5fc8a49d00
[    5.371802] R13: ffff9b5fbb4c2cc0 R14: ffff9b5fc85940a0 R15: ffff9b5fc7930000
[    5.371804] FS:  00007f0207847940(0000) GS:ffff9b5fce340000(0000) knlGS:0000000000000000
[    5.371805] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    5.371806] CR2: 00007f194287f910 CR3: 000000013bac8002 CR4: 00000000000606e0
[    5.371807] Call Trace:
[    5.371811]  sysfs_create_dir_ns+0x77/0x90
[    5.371815]  kobject_add_internal+0xac/0x2b0
[    5.371819]  ? kfree_const+0x20/0x30
[    5.371820]  kobject_add+0x71/0xd0
[    5.371822]  ? kobject_create+0x1b/0x40
[    5.371824]  kobject_create_and_add+0x31/0x60
[    5.371827]  pci_init_one+0x323/0x910 [men_lx_chameleon]
[    5.371830]  ? _cond_resched+0x19/0x40
[    5.371833]  local_pci_probe+0x47/0xa0
[    5.371836]  pci_device_probe+0x145/0x1b0
[    5.371839]  driver_probe_device+0x31e/0x490
[    5.371841]  __driver_attach+0xa7/0xf0
[    5.371843]  ? driver_probe_device+0x490/0x490
[    5.371845]  bus_for_each_dev+0x70/0xc0
[    5.371847]  driver_attach+0x1e/0x20
[    5.371849]  bus_add_driver+0x1c7/0x270
[    5.371851]  ? pci_init_one+0x910/0x910 [men_lx_chameleon]
[    5.371853]  driver_register+0x60/0xe0
[    5.371855]  ? pci_init_one+0x910/0x910 [men_lx_chameleon]
[    5.371856]  __pci_register_driver+0x5a/0x60
[    5.371858]  men_chameleon_init+0x60/0x90 [men_lx_chameleon]
[    5.371861]  do_one_initcall+0x52/0x19f
[    5.371863]  ? _cond_resched+0x19/0x40
[    5.371865]  ? kmem_cache_alloc_trace+0xa6/0x1b0
[    5.371868]  ? do_init_module+0x27/0x209
[    5.371870]  do_init_module+0x5f/0x209
[    5.371872]  load_module+0x191e/0x1f10
[    5.371874]  ? ima_post_read_file+0x96/0xa0
[    5.371877]  SYSC_finit_module+0xfc/0x120
[    5.371878]  ? SYSC_finit_module+0xfc/0x120
[    5.371881]  SyS_finit_module+0xe/0x10
[    5.371883]  do_syscall_64+0x73/0x130
[    5.371886]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
```
The problem is that men_chameleon.c is using only the name of the FPGA from the chameleon table to create the sysfs entry. Therefore a incrementing number or an unique identifier is needed in addition to create the sysfs entry.

mlassota, 2019-04-01:
Could you please provide more info about:
- the hardware directly involved (which two devices)
- how drivers are loaded (i.e. what params)
- any extra commands called

I'd like to replicate the issue you have.

AndreasGeissler, 2019-04-01:
Hello Marcin,

I have seen this issue the first time with a F21P and 2x CC10 cards when trying to load the men_lx_z77 driver on F21P. I didn't used any additional params and no other commands.

This issue will always occur when having two FPGA's in the system with an identically FILENAME. The filename is used by the men_chameleon.c to create a sysfs entry and the kernel will complain when the same name is used twice. Since this is related to men_chameleon.c only native Linux drivers will be affected.

check the following line in the driver:
h->chamTblObj = kobject_create_and_add( tblfile,  &G_cham_devs->kobj );
were tblfile name is the name of the FPGA.

Maybe this is also related to the kernel version.

Regards,
Andreas

mlassota, 2019-04-03:
I've fixed this by prefixing directory name with index.
Also I've trimmed leading spaces of FPGA file name to not create a lame directory names (e.g. "02_&nbsp;&nbsp;&nbsp;16F614-00" => "02_16F614-00").

closed 2019-04-03 by mlassota