MEN-Mikro-Elektronik/13MD05-90 #68: Integrate MT LL driver pkg into 13MD05-90 as submodule

dpfeuffer, 2019-01-14:
Please integrate MT LL driver pkg into 13MD05-90 as submodule like other LL drivers.
See https://github.com/MEN-Mikro-Elektronik/mdis_ll_mt

dpfeuffer, 2019-01-15:
Please wait with this issue because two MT tools are missing! I will update it soon.

dpfeuffer, 2019-01-16:
The supposed missing test tools are Windows specific and are no longer part of the MT LL package. I updated the documentation under DOC and removed the description of the two tools.
The next days, I will rewrite the README.md of mdis_ll_mt.

mlassota, 2019-04-24:
The MT_CALLBACK tool does not compile:
~~~~
++++++++ Building mt_callback +++++++++++
Directory OBJ/mt_callback created
Compiling mt_callback.c
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c: In function ‘main’:
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c:204:2: error: unknown type name ‘UOS_CALLBACK_HANDLE’
  UOS_CALLBACK_HANDLE *cbHdl;
  ^
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c:286:15: warning: implicit declaration of function ‘UOS_CallbackInit’ [-Wimplicit-function-declaration]
  if ((error = UOS_CallbackInit(path, &cbHdl))) {
               ^
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c:302:16: warning: implicit declaration of function ‘UOS_CallbackClear’ [-Wimplicit-function-declaration]
   if ((error = UOS_CallbackClear(cbHdl, 100))) {
                ^
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c:313:16: warning: implicit declaration of function ‘UOS_CallbackSet’ [-Wimplicit-function-declaration]
   if ((error = UOS_CallbackSet(cbHdl, 0, CallbackFkt_0, (void*)0xaaaa))) {
                ^
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c:416:5: warning: implicit declaration of function ‘UOS_CallbackMask’ [-Wimplicit-function-declaration]
     UOS_CallbackMask(cbHdl);
     ^
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c:421:5: warning: implicit declaration of function ‘UOS_CallbackUnMask’ [-Wimplicit-function-declaration]
     UOS_CallbackUnMask(cbHdl);
     ^
/opt/menlinux/DRIVERS/MDIS_LL/MT/TOOLS/MT_CALLBACK/COM/mt_callback.c:477:16: warning: implicit declaration of function ‘UOS_CallbackExit’ [-Wimplicit-function-declaration]
   if ((error = UOS_CallbackExit(&cbHdl)))
                ^
~~~~

If this tool is required please provide proper library as I can't find it on github.

dpfeuffer, 2019-04-24:
Only the SW modules specified in mt.xml must be buildable.
See https://github.com/MEN-Mikro-Elektronik/mdis_ll_mt/blob/master/PACKAGE_DESC/mt.xml

Background: MDIS specifies callback functionality optional. MDIS for Linux doesn't support callbacks (see https://github.com/MEN-Mikro-Elektronik/13MD05-90/blob/master/MDISforLinux/INCLUDE/NATIVE/MEN/usr_os.h : #define NO_CALLBACK).

mlassota, 2019-04-24:
Would it be OK if I remove MT_CALLBACK tool from repository? It's not in mt.xml after all.

dpfeuffer, 2019-04-24:
No, please keep it, because it may be usable under QNX or VxWorks in the future. But then, this tool must be excluded for Linux. Please try the following:
Add the MT_CALLBACK tool to the mt.xml and add the tag attribute notos:
```
<swmodule internal="true">
    <name>mt_callback</name>
    <description>MDIS callback test tool</description>
    <type>Driver Specific Tool</type>
    <makefilepath>MT/TOOLS/MT_CALLBACK/COM/program.mak</makefilepath
    <notos>Linux</notos>        
    <notos>Windows</notos>        
</swmodule>
```
Then please verify that mdiswiz considers the notos tag (that the tool is excluded from the sw module list).

mlassota, 2019-04-24:
I've added mt_callback to the mt.xml like you said.
mdiswiz deals with notos tag with no problem.

closed 2019-04-24 by mlassota