MEN-Mikro-Elektronik/13MD05-90 #69: failure during make process, Kernel 4.14.93

M-Gerner, 2019-01-17:
Using a Ubuntu 18.04 with kernel 4.14.93-041493-generic I tried to make a project with a F26L, a F205 and a M99.
It was always stopped at the same point of the buildprocess.
[image]
[buildprocess.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2769320/buildprocess.txt)

At first I removed the the m99 and the F205 from the project.
I tried this process with the latest version of the developers branch on github as well as the master branch and a older mdis version (13mdis05-90_01_20).

[content_makefile.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2769366/content_makefile.txt)
[content_systemdsc.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2769367/content_systemdsc.txt)

JSierl, 2019-01-22:
probably a duplicate/same error as handled during pull request #67

KonradJPE, 2019-03-06:
fixed by merge https://github.com/MEN-Mikro-Elektronik/13MD05-90/pull/67

closed 2019-03-06 by KonradJPE