MEN-Mikro-Elektronik/13MD05-90 #74: F26L cannot write/read from M77N on G204 board.

KonradJPE, 2019-02-11:
No data can be received or send via UART on M77N while  attached to G204 board.

M77 is detected correctly, as well as modprobing men_lx_m77 is successful. 
The main difference with working solution on F23P is in lspci log and corresponds to IRQ number.

F23P:
04:0d.0 Bridge: Altera Corporation Device d203 (rev e3)
    Subsystem: Device ff00:ff00
    Flags: 66MHz, medium devsel, IRQ 10
    Memory at b8201000 (32-bit, non-prefetchable) [size=4K]
    Kernel modules: altera_cvp
   
F26L:
0e:0d.0 Bridge: Altera Corporation Device d203 (rev e3)
    Subsystem: Device ff00:ff00
    Flags: 66MHz, medium devsel, IRQ 255
    Memory at 9b601000 (32-bit, non-prefetchable) [size=4K]
    Kernel modules: altera_cvp

While booting from UEFI Memory at 9b601000 on F26L is disabled by default, but it can be changed via command
sudo setpci -s 0e:0d.0 COMMAND=x 
[F23P_lspci.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2851330/F23P_lspci.log)
[F26L_lspci_after_boot.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2851331/F26L_lspci_after_boot.log)
[F26L_lspci_after_setting_pci_COMMAND.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/2851332/F26L_lspci_after_setting_pci_COMMAND.log)

JSierl, 2019-03-06:
Probably a HW bug

JSierl, 2019-03-28:
There will be a new F26L BIOS (approximately May 2019)
The Issue can then be tested again

KonradJPE, 2019-10-01:
F26L with M35 on G204 fails interrupt test. 
`sudo m34_blkread m35_1 -r=14 -b=1 -i=3 -d=1 `
Internal MEN PR number: PR005764

Issue still persists.

dpfeuffer, 2019-10-02:
I will send you an e-mail about the upgrade of your F26L to the newest UEFI version. Then please verify, if the problem is solved.

KonradJPE, 2019-10-02:
After updating UEFI to version 2.04 above problems does not occur.

dpfeuffer, 2019-10-07:
KonradJPE reported us per mail:
>G204 are detected and working correctly with attached M-Modules only when corresponding memory regions are enabled.
This still has to be set manually by command: 
'setpci -s xx.xx.xx COMMAND'

So we have still a problem with this HW combination. It shall work without any manual workaround.
Our UEFI developer, Alfred Jendrusch, will perform some tests this week.
However, we will ship you a new F26L with new UEFI for further testing (after official release of the new UEFI).

dpfeuffer, 2019-11-15:
After internal discussion, we decided to solve the problem by enabling the memory regions within all affected PCI related drivers.
I will investigate which drivers are affected, add the code to enable the memory regions, make a basic test (at least with one driver) and provide you the changes via pull request(s) ASAP. The modifications shall be considered for current release 2.01.

dpfeuffer, 2019-11-22:
Which linux distro did you use for the tests?

I'm using CentOs 8 UEFI installation with current 13MD05-90 jpe-dev branch and G25A+G204+M99.
After power up, the mem region of G204 FPGA is disabled.
After first "m_open m99_1", the mem region is enabled, and I get the following message from dmesg:
```pci 0000:07:00.0: enabling device (0100 -> 0102)```
m99_latency works without problems.
I cannot find the string "enabling device" within /opt/menlinux.
When I manually disable the mem region "setpci -s 07:00.0 COMMAND=0100", m99_latency shows no longer interrupts.
After reloading the m99 driver with "rmmod men_ll_m99" and "modprobe men_ll_m99" the mem region is still disabled.

So some SW instance in my system enables the mem region of the G204 FPGA.
For details, see attached file.
[CentOS_UEFI_G25A_G204_M99.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3880339/CentOS_UEFI_G25A_G204_M99.txt)

dpfeuffer, 2019-11-25:
>After reloading the m99 driver with "rmmod men_ll_m99" and "modprobe men_ll_m99" the mem region is still disabled.

The mem region is still disabled after reloading of all men drivers and "m_open m99_1".

dpfeuffer, 2019-11-25:
CentOs 7.5 legacy installation with G25A+G204+M99.
After power up and linux boot:

```
[root@mdis4lin ~]# setpci -s 07:00.0 COMMAND
0107
```

So G25A UEFI CSM boot enables mem+io mapped region and sets bit 2 (Bus Master). This bit is not set with CentOs 8 UEFI installation boot and  first "m_open m99_1.

KonradJPE, 2019-11-25:
@dpfeuffer Please see my comments: 

> Which linux distro did you use for the tests?

Mainly Ubuntu.  For below tests I was using Ubuntu 18.04.2 LTS

> After first "m_open m99_1", the mem region is enabled, and I get the following message from dmesg:
> pci 0000:07:00.0: enabling device (0100 -> 0102)

I do not have m99_1 M-Module, but it is possible to reproduce the results using other M-Modules (I used m11 and m82)

> I cannot find the string "enabling device" within /opt/menlinux.

This log is produced by kernel pci_enable_device function, which is invoked by OSS_PciGetConfig in LIBSRC/OSS/oss_bustoaddr.c MDIS. 

> When I manually disable the mem region "setpci -s 07:00.0 COMMAND=0100", m99_latency shows no longer interrupts.
> After reloading the m99 driver with "rmmod men_ll_m99" and "modprobe men_ll_m99" the mem region is still disabled.
> The mem region is still disabled after reloading of all men drivers and "m_open m99_1".

This behaviour is consistent with the description of the pci_enable_device function. Please see https://www.kernel.org/doc/htmldocs/kernel-api/API-pci-enable-device.html

> G25A UEFI CSM boot enables mem+io mapped region and sets bit 2 (Bus Master).

I am wondering if COMMAND register value during scan_system.sh script is set correctly. 
Now this value is fixed - we are setting Memory Space and I/O Space (value 0x3).

KonradJPE, 2019-11-25:
Update: 
> I am wondering if COMMAND register value during scan_system.sh script is set correctly.
> Now this value is fixed - we are setting Memory Space and I/O Space (value 0x3), if we detect that memory regions are disabled.

I can see your the answer in other issue, thanks.
https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/162

dpfeuffer, 2019-11-25:
>This log is produced by kernel pci_enable_device function, which is invoked by OSS_PciGetConfig in LIBSRC/OSS/oss_bustoaddr.c MDIS.

Yes, I found this morning the "enabling device (%04x -> %04x)\n" in pci_enable_resources(), invoked from pcibios_enable_device() <-- do_pci_enable_device() <-- pci_enable_device_flags() <-- pci_enable_device().

I did a grep pci_enable_device but found only matches in drivers but not in OSS. Now I found the reason: My MDISforLinux/LIBSRC/OSS folder was empty because of the move of OSS into a submodule. A "git submodule init" and "git pull --recurse-submodules" fixed it now.

Very interesting that OSS_PciGetConfig() calls pci_enable_device() !
**Obviously most of our PCI related MDIS drivers calls OSS_PciGetConfig(), therefore I see now no longer the necessity to modify the PCI drivers as planned.**

>This behaviour is consistent with the description of the pci_enable_device function. 
Yes, you are right, I read this already with the counter, but I didn't see the conjunction with this behavior. 

@KonradJPE 
You wrote in e-mail from 2019-10-07:
>G204 are detected and working correctly with attached M-Modules only when corresponding memory regions are enabled.
This still has to be set manually by command: 
'setpci -s xx.xx.xx COMMAND'

Now, I assume "detected" isn't possible without setpci.
However, "working correctly" did not require setpci, because men_bb_d203 calls OSS_PciGetConfig() which calls pci_enable_device(). 

**CONCLUSION**
No further modifications in PCI drivers necessary to enable mem regions, because:
- DRIVERS/CHAMELEON/men_chameleon.c calls pci_enable_device() explicitly
- other PCI drivers calls OSS_PciGetConfig() which calls pci_enable_device()

Please reopen this problem if you do not agree with my conclusion.

closed 2019-11-25 by dpfeuffer