MEN-Mikro-Elektronik/13MD05-90 #76: All tools should be buildable from source.

KonradJPE, 2019-02-12:
No pre delivered binaries in MDIS shall be provided.  
All tools should be buildable from source - as fpga_load is now.

dpfeuffer, 2019-02-12:
What about the MDIS Wizard? We have to provide the binary in 13MD05-90 because it is the starting tool for the customers to scan/configure a target system. But, because 13MD05-90 is distributed under GPL2, all the sources to build the mdiswiz binary must be included (or delivered on request?). As far as I know, a distinct and patched QT lib source code is required, but I don't know if we have the permission to include this patched QT lib sources as well. Tbd.

KonradJPE, 2019-03-15:
Could you please confirm if below solution is acceptable? 
After installing MDIS sources via INSTALL script user will be asked to build MDIS tools.
Below tools will be build: 
-fpga_load
-mm_ident
-(others ?) - please specify 

In order to achieve this mm_ident tool have to be added into 13MD05-90 repostory as submodule. 
Do you agree?

dpfeuffer, 2019-03-15:
The following tools shall be built optional:
- fpga_load
- mm_ident

mm_ident shall be added as submodule.

I would prefer to build the optional tools not during the MDIS installation (at the end of INSTALL script). If mm_ident is used from scan_system.sh the tool must be built prior the scan.

How about building the optional tools stand alone using their own makefiles? e.g.
cd fpga_load
make

cd mm_ident
make

dpfeuffer, 2019-03-26:
The requirement about binary files was changed from:
```
<MDIS_ConfRule_0014>:
No binaries or executables shall be part of any repository
```
To
```
<MDIS_ConfRule_0014> Binary Constrain
For all included binaries of any repository, the belonging sources must be available publicly.
```
In https://github.com/MEN-Mikro-Elektronik/13MD05-90/wiki/Configuration-Management-Plan-for-MDIS/cfa8ba3476c68dbf93a06f43f7a9975037ac2cb9

fpga_load and mm_ident shall be standalone buildable (using make) for the target architecture. For further information please refer to the MDIS Install Script Requirements (currently in work).

KonradJPE, 2019-04-29:
> The following tools shall be built optional:
>     fpga_load
>     mm_ident

Tools will be build when architecture is different  than "86". Furthermore it is possible to build fpga_load and mm_ident when `--buildtools` is passed to scan_system.sh script. 
`sudo /opt/menlinux/scan_system.sh /opt/menlinux --buildtools` 

It is also possible to build tools manually in MDIS installation dir, like:
```
cd /MDIS_INSTALLATION_DIR/TOOLS/FPGA_LOAD
make

cd /MDIS_INSTALLATION_DIR/TOOLS/mm_ident
make
```
In case of error or unexpected behaviour of scan_system.sh script please create new issue.

closed 2019-04-29 by KonradJPE