MEN-Mikro-Elektronik/13MD05-90 #78: System hangs while running command 'mdis_createdev' on F23P with M77

KonradJPE, 2019-02-25:
System hangs while running command 'mdis_createdev' on F23P with M77 - Exception while running interrupt.
**Issue occurs in configuration:**
CompactPCI:
-F205 (M31,M66)
-F215
CompactPCI Serial (starting from F23P + F614)
-Free slot
-G204 with M77
-G215

**When the system configuration is changed (same cards, no free slots) everything works.**
CompactPCI:
-F205 (M31,M66)
-F215
CompactPCI Serial (starting from F23P + F614)
-G204 with M77
-G215

JSierl, 2019-03-06:
Probably M77N problem. We will update this modules when current milestone 
13MD05-90_02_00 is completely closed

JSierl, 2019-03-28:
Please send your M77N Modules to us so we can update them.

KonradJPE, 2019-06-17:
After updating M77 issue does not occur. 
**Tested configurations:**
**Os:** 
Kernel 4.15.0-46 - Ubuntu 18.04.02
Kernel 4.14.35, 4.10.17 -Ubuntu 16.04.05 LTS 64bit.
**Hardware:**
**CPU F23P**
CompactPCI:
-F205 (M31,M66)
-F215
CompactPCI Serial (F23P with F614)
-Free slot
-G204 with M77
-G215

**CPU G22**
CompactPCI Serial (G22)
-Free slot (3x)
-G215
-Free slot
-G204 with M77

closed 2019-06-17 by KonradJPE