MEN-Mikro-Elektronik/13MD05-90 #81: Remove deprecated Makefiles from repository.

KonradJPE, 2019-02-25:
Below Makefiles, and dependences should be deleted from 13MD05-90 repostitory. 

-    13MD05-90/13AD78-06/AD78/EXAMPLE/AD78_WIN/COM/program.mak

- 13MD05-90/13M062-06/DRIVERS/MDIS_LL/M062/EXAMPLE/M62_LL/COM/program.mak

- 13MD05-90/13M066-06/DRIVERS/MDIS_LL/M066/EXAMPLE/M66_LL/COM/program.mak

- 13MD05-90/13Y004-06/SMB2/TOOLS/SMB2_BMC_CTRL/COM/program.mak

- 13MD05-90/MDISForLinux/LIBSRC/UTI/library.mak

dpfeuffer, 2019-02-25:
I agree to:
- AD78_WIN
- M62_LL
- M66_LL
But why would you like to remove:
- SMB2_BMC_CTRL: I revised the tool for the current MEN BMC 2018-12, I think it's useful
- UTI: Is this lib not used from any linux tool?

mlassota, 2019-02-26:
SMB2_BMC_CTRL tool does not compile on Linux, because there is no smb2_bmc_api library nowhere in 13MD05-90 repository nor in the other repositories on GitHub. You have to provide smb2_bmc_api library.

mlassota, 2019-02-26:
UTI library is not used on Linux in 13MD05-90.

dpfeuffer, 2019-02-27:
SMB2_BMC_CTRL:
See the open pull request for 13Y004-06, which includes the smb2_bmc_api and some modifications of SMB2_BMC_CTRL:
https://github.com/MEN-Mikro-Elektronik/13Y004-06/pull/1

UTI:
ok, then please remove it

KonradJPE, 2019-03-21:
Removed: 
`AD78_WIN`
`M62_LL`
`M66_LL`
Updated:
`SMB2_BMC_CTRL`
`QSPIM_CB`
`UTI`

KonradJPE, 2019-03-21:
Should the doxygen documentation be generated again for affected submodules?

dpfeuffer, 2019-03-22:
I think you are asking because of the results in 13MD05-90_SwDeliveryAcceptanceReport.
Please update the documentation when you working on submodules. But if the work is already done, you don't have to update the documentation now.
See new issue https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/95

closed 2019-03-22 by KonradJPE