MEN-Mikro-Elektronik/13MD05-90 #84: INSTALL script shall copy changelog into /opt/menlinux dir

dpfeuffer, 2019-02-27:
jpe-dev 386e523

INSTALL script shall copy 13MD05-90_changelog.md into /opt/menlinux.

dpfeuffer, 2019-03-27:
Here are some requirements for the INSTALL script and scan_system.sh script:
Note: Use a markdown editor with mermaid support to view the flow charts, or just open the PDF.

[MDIS Install and Scan Script Requirements.md.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3013685/MDIS.Install.and.Scan.Script.Requirements.md.txt)

[MDIS Install and Scan Script Requirements.pdf](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3013661/MDIS.Install.and.Scan.Script.Requirements.pdf)

KonradJPE, 2019-04-25:
Additionally to 

> MDIS Install and Scan Script Requirements.pdf

scan_system script should be more user friendly. Additional options like: 
--verbosity
--PCIdrytest
--build_tools
--help 
should be provided.

Moreover: 

> scan_system.sh requires always the mdis_installation_path as first argument: “scan_system.sh /opt/menlinux”. The script shall be compatible for this call. And I suggest, without any argument, the script should show a proper usage information (purpose of the script, possible required/optional arguments).

KonradJPE, 2019-04-29:
> scan_system script should be more user friendly.

Added new options for scan_system.sh script (help output below):

```
scan_system.sh   script to generate an automatic MDIS configuration
                 when doing a selfhosted project"

  The script checks which CPU we are on. The SMB2 drivers
  are always added to the CPU to provide BMC and watchdog
  drivers.

IMPORTANT:
     1. Path to MDIS install dir has to passed as first argument
     2. Tools must be available in system: i2cdump, setpci, lspci,
        libelf-dev or libelf-devel."
     3. /usr/src/linux/ points to valid kernel headers

parameters:
     'MEN_LIN_DIR'     path to MDIS installation dir - default
                       /opt/menlinux/"
     --verbose         if 1 or 2 then additional debug info is dumped
                       ex: --verbose 1
     --buildtools      build mm_ident and fpga_load from source
     --drytest         run drytest with test PCI device list 
                       if passed as alternative PCI devices temp"
                       file the default file /tmp/men_pci_devs is
                       not written. Used to test system.dsc
                       generation with predefined test data.
                       ex: --drytest sth
     --prerequisites   check if all prerequisites are met
     --help            print help
```

KonradJPE, 2019-05-02:
Almost all requirements are met, known issues: 
1. fpga_load and mm_ident compilation warnings. 
Issue:  https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/111
2. Check if libelf-dev, libelf-devel or elfutils-libelf-deve is required during MDIS build:
"Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
3. scan_system.sh shall support M-Module detection of all M-Modules (common but not M-Module specific implementation)
Issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/85

KonradJPE, 2019-07-04:
1. Fixed.
2. Added information into scan_system.sh prerequisites.

> In case of warning: "Cannot use CONFIG_STACK_VALIDATION=y please install libelf-dev, libelf-devel or elfutils-libelf-devel".

3. Fixed.

closed 2019-07-04 by KonradJPE