MEN-Mikro-Elektronik/13MD05-90 #85: Add support for all M-Modules available in MDIS repository

KonradJPE, 2019-03-01:
Create grep+regex+awk xml parser, that will obtain M-Module data from xml files and will add M-Module entity into system.dsc. Morover proper .mak files will be added into main Makefile.

mlassota, 2019-04-04:
Only M31, M35, M36N, M66, M72, M77 and M82 are currently supported cause these are the only M-Module we have and so can test.

dpfeuffer, 2019-04-05:
scan_system.sh shall handle all M-Modules without the need of M-Module specific code within the script or M-Module specific template files. This should be possible at least for M-Modules where MDIS low-level drivers (non native drivers like for M77) are used.

A few notes (base on jpe-dev f7a3e2e6104cf67eff0c0e2a120bf95ab1991be9):
- One M-Module descriptor template (mX.tpl) should be sufficient
- DEBUG_LEVEL_XXX entries could be omitted because the default 0xc0008000 are used from the drivers when now entry is set.
- mdiswiz parses all .xml files at the beginning to know the relationship of ll-drv to HW-ids.
- Each .xml file of M-Module drivers contains the M-Module Id (<mmoduleid>) programmed in the M-Module EEPROM. When an M-Module carrier is found, then all .xml files could be scanned for <mmoduleid> to create a .xml-file to <mmoduleid> table/array. For each found M-Module at the carrier, the proper .xml file can be parsed to determine:
  - HW_TYPE from the driver name e.g. <name>m34</name> => HW_TYPE=M34
  - _WIZ_MODEL from <hwname>, e.g. <hwname>M35</hwname> => _WIZ_MODEL=M35
- For a few M-Modules (e.g. CAN) more then one driver package with .xml is available. I think it is ok to use just the first found .xml (driver) instead of asking the user which one to use. It is really seldom and the customer can change which driver to use later on with mdiswiz.

mlassota, 2019-04-15:
It all comes to having a good XML parser. As we agreed earlier we won't be using any tool beside bash, grep, sed and awk. It makes it difficult to parse because XML documents hardly submit to regex parsing. What we need is to write XML parser in bash. So my concept is to write SAX parser as it seems to be the fastest and requires less memory footprint. But it will take some time.
Or maybe you have an other idea?

dpfeuffer, 2019-04-23:
I did a little bit XML parsing of MDIS package desc files in the check_pkgdesc.sh script. See https://github.com/MEN-Mikro-Elektronik/mdis_tools_dev. Maybe it is helpful for you.

mlassota, 2019-04-23:
What you have written there is actually a kind of sequential parser. The problem is that if you want to parse a XML file with this parser you need not only a valid XML file but also a valid XML file for this parser.
To get a better picture let's see the #104 issue. The script claims that there's invalid entry in XML file:
``` --- 13z02406.xml: MDIS5 driver package for Z24 SRAM - MMODPROG --- ```
``` *** <docroot>=DRIVERS/MDIS_LL/MMODPRG/DOC/html/index.html not found ```
But as a matter of fact this entry is commented (Konrad removed this line anyway [2244259](https://github.com/MEN-Mikro-Elektronik/13Z024-06/commit/224425996a92c93f77c90685de6ffa04fb77f9ad#diff-8c8fbe9d54c5b6a71038f649005eaa69)):
```  ```
Therefore it should be ignored just like all other comments are.
So the script is good, but not that good. It needs to be improved. But it will require more time to work on it, as there probably will be more cases to consider apart from the example I've given.
So my point is that you can accept script with such a flaws or we can spend more time to improve this.
It's up to you to decide.

dpfeuffer, 2019-04-24:
I agree with you that check_pkgdesc.sh is not perfect and I had overlooked the fact, that docroot in 13z02406.xml was just a comment. I was just about to consider this case in the script, but now I think commented out xml tags should not be in package desc files that are delivered to customers. So this flaw is now a feature for me. ;-) Thanks for the removal of this comment line. 
Only valid package desc xml files shall be delivered to customers, so it is ok to rely on valid xml files only (and commented out tags must not exist). So I decide that such flaws are accepted by us and you can write the script more straight forward.
I'm thinking about to improve check_pkgdesc.sh to warn about commented out tag lines in general.

mlassota, 2019-04-24:
OK. Thank you for your explanation :)

KonradJPE, 2019-04-29:
After creation of xml parser, Makefile output from mdiswiz and scan_system script should be similar. 
Additional information can be found in point 4 of issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/62

mlassota, 2019-06-14:
This is done.

When M-Module is detected for the first time by scan script, a M-Module ID to XML file mapping array is created, i.e. all 13m0*.xml files are parsed, thus it may take some time.
Next a matching XML file is parsed and proper M-Module section is created in system.dsc file based on data from XML file.

closed 2019-06-14 by mlassota