MEN-Mikro-Elektronik/13MD05-90 #89: *.c files may include tested kernel version

FlorianRauh, 2019-03-14:
For example the "men_16z077_eth.c" includes old tested kernel versions.
Please remove all comments in .c files which you have tested

/*********************  P r o g r a m  -  M o d u l e ***********************/
/*!
 *        \file  men_16z077_eth.c
 *      \author  thomas.schnuerer@men.de
 *
 *        \brief driver for IP core 16Z087 (Ethernet cores).
 *               supports kernel 3.0 to 4.8
 *
 *     Switches:
 *
 *---------------------------------------------------------------------------
 * Copyright (c) 2019, MEN Mikro Elektronik GmbH
 ****************************************************************************/

KonradJPE, 2019-03-15:
> Please remove all comments in .c

As far as I understand, we should remove only comments, that specify the supported kernels and "\brief" section. NOT the all comments in .c files? 
> which you have tested

We have compiled all available makefiles which cover almost all .c .h files. Do you mean functional tests?

dpfeuffer, 2019-03-27:
For clarification:
No source files within 13MD05-90 shall contain information under which kernel versions the module was tested, because this information may be outdated. 
Information about restrictions or special requirements for certain kernel versions are ok.
@FlorianRauh: Do you agree?

JSierl, 2019-04-25:
> 
> 
> For clarification:
> No source files within 13MD05-90 shall contain information under which kernel versions the module was tested, because this information may be outdated.
> Information about restrictions or special requirements for certain kernel versions are ok.
> @FlorianRauh: Do you agree?

I discussed with Florian and this is absolutely correct.

KonradJPE, 2019-04-29:
It seems that only module 13Z077-90 was affected.
I could not find any other module that contain information about supporting specific kernel version.

closed 2019-04-29 by KonradJPE