MEN-Mikro-Elektronik/13MD05-90 #90: driver version of 13Z077-90

FlorianRauh, 2019-03-14:
Dmesg is telling me when I load the 13Z077-90 driver that I´m using version 1.45
this is fixed string in men_16z077_eth.c which is not up to date.
`static const char *version = "$Id: men_16z077_eth.c,v 1.45 2014/07/16 19:30:45 ts Exp $";`
I´m pretty sure that updates have been taken place between 2014/07/16 and now.

dpfeuffer, 2019-03-27:
We have to evaluate how to support revision info in general. tbd.

JSierl, 2019-04-25:
I am going to evaluate this together with Dieter within the next week.

dpfeuffer, 2019-05-06:
The driver version of 13Z077-90 shall be handled like the requirements defined in https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/94.

KonradJPE, 2019-06-10:
Fixed.

closed 2019-06-10 by KonradJPE