MEN-Mikro-Elektronik/13MD05-90 #91: 13MD05-90_02_00_b1 release: all submodule packages have wrong tags

dpfeuffer, 2019-03-15:
All submodules referenced from 13MD05-90_02_00_b1 are tagged with the same tag revision 02_00_b1 of 13MD05-90 but must be tagged with it's own submodule specific tag.

Example 13M036-06:
[image]

Last tag was: 13M036-06_02_05
Next tag should be: 13M036-06_02_06

Please check all submodules.

KonradJPE, 2019-03-15:
Layout: <name>_<major-rev>_<minor_rev> 
For all submodules should we change only <minor_rev> ?

dpfeuffer, 2019-03-15:
The revision tags of the submodules shall be completely independent from the revision tag of the 13MD05-90 package.
The submodules may be used from other repos (MDIS for QNX in the future).
Only when a submodule was changed, a new tag mussed be assigned and this tag shall be based on the last tag of this submodule.
Example:
minor change: 01_04 -> 01_05
bigger change: 01_04 -> 02_00

KonradJPE, 2019-03-15:
It seems, that the same mistake have been done while creating release 01_21. 
I am planning to solve this in the following way: 
1. Remove tags 01_21, 02_00_b1 from submodules,
2. Create new tags, that will correspond commit from 01_21, and 02_00_b1 release. 
So 2 tags will be added to submodules. Please see 13M036-06 as example:
13M036-06_01_21 -> 13M036-06_02_06
13M036-06_02_00_b1 ->  13M036-06_02_07

Could you please confirm?

KonradJPE, 2019-03-18:
What about submodules, that do not have any tags defined for now?
In this case, should the tag corresponds the MDIS release version?

dpfeuffer, 2019-03-18:
>Could you please confirm?
Yes, if the 13M036-06 submodule was changed between 13MD05-90 releases 01_20 and 01_21 and between 13MD05-90 releases 01_21 and 02_00.

>What about submodules, that do not have any tags defined for now?
>In this case, should the tag corresponds the MDIS release version?
Which submodules?
The first tag of a submodule should reflect the last MCVS version +1.
Example: If a submodule was rev. 1.14 in MCVS, then the first Git tag should be _01_15.

KonradJPE, 2019-03-20:
> What about submodules, that do not have any tags defined for now?
> In this case, should the tag corresponds the MDIS release version?
> Which submodules?
> The first tag of a submodule should reflect the [...]

We do not have access to MCVS, so we cannot check the version. Currently tags are fixed for all 13* submodules. Could you please provide the proper version for submodules in

dpfeuffer, 2019-03-20:
You will find the MCVS versions in the (removed) history.txt files of the submodule repos. The last versions may be manual edited entries because the migration to git. Please find out the last commit where the history.txt file is not removed.
Example for mdis_drivers_bbis_a12_com:
https://github.com/MEN-Mikro-Elektronik/mdis_drivers_bbis_a12_com/blob/772a3647aac1009a4e5af7088f54740422e253f0/a12_history.txt
-----[ Revision 1.33 ]-----
Because there are some commits after this last commit with history.txt, I suggest to use _1_43 (old-rev + 0.10) as tag name (not "last MCVS version +1" what I recommended two days ago). With this, we have an indicator that some revision numbers are lost between last MCVS revision and the current tag.

dpfeuffer, 2019-03-21:
Beta tags for submodules make only sense for a beta release of the submodule itself, but not when 13MD05-90 will be released with a beta tag for the acceptance test.

KonradJPE, 2019-03-21:
> Please find out the last commit where the history.txt file is not removed.

Several submodules does not contain history file at all. I have tried to get revision number from .c files, but each file seems to have different rev number. Is it possible for you to obtain correct revision for below submodules? 
mdis_include_com_men
mdis_libsrc_mbuf_com
mdis_libsrc_usr_utl_com
mdis_tools_mdis_api
mdis_drivers_bbis_a21_com (history file without revision)
mdis_drivers_bbis_smbpci_com
mdis_tools_wdog
mdis_tools_m_tmr
mdis_tools_fpga_load

dpfeuffer, 2019-03-21:
Please use the following revisions for the current tags:

mdis_include_com_men : _02_00
mdis_libsrc_mbuf_com : _02_26
mdis_libsrc_usr_utl_com : _01_41
mdis_tools_mdis_api : _01_41
mdis_drivers_bbis_a21_com : _01_11
mdis_drivers_bbis_smbpci_com : _02_44
mdis_tools_wdog : _02_10
mdis_tools_m_tmr : _01_16
mdis_tools_fpga_load : _01_21

KonradJPE, 2019-03-21:
There is only one submodule left. 
What should be the correct tag for pciutils?  I have created pciutils_03_05_06, but this not meet the requirement \<name\>\_\<major-rev\>\_\<minor\_rev\>. 
https://github.com/MEN-Mikro-Elektronik/pciutils/releases

dpfeuffer, 2019-03-22:
Good question ;-)
pciutils is not from MEN and has it's own revision schema. It looks like, that our fork is based on release 3.5.5. Because the based release version should be visible, I suggest in this case to use the official version info + a prefix: pciutils_<official-version>_<MEN-version-prefix>

Because I see three dates (with four commits) after "Released as 3.5.5" commit, I would use:
pciutils_3.5.5_03 as tag for the current commit.

KonradJPE, 2019-03-22:
Submodules tags fixed.

closed 2019-03-22 by KonradJPE