MEN-Mikro-Elektronik/13MD05-90 #93: mdiswiz produces fonrconfig errors/warning under CentOs 7

dpfeuffer, 2019-03-20:
Linux mdis4lin 3.10.0-862.el7.x86_64 #1 SMP Fri Apr 20 16:44:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

When I start mdiswiz under CentOs 7, the following errors/warnings appear. Please check if it is possible to fix this minor issue. These problem may not seem to influence the mdiswiz behaviour.

```
root@mdis4lin mproj_wizscan]# /opt/menlinux/BIN/mdiswiz 
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 70: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 70: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 78: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/65-0-lohit-bengali.conf", line 32: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-sans.conf", line 24: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-serif.conf", line 24: unknown element "langset"
```

mlassota, 2019-04-04:
This issue rather affects all new distros. That's because MDIS Wizard (Qt, to be exact) is using old fontconfig static library (version 2.8.0) and the configuration files it tries to read come from the machine it is running on and these file usualy come from newer fontconfig version, thus they can't be read properly (no forward compatibility).

There's no way to disable this on fontconfig side with some kind of switch. The only way would be to disable it in source code, but this is not the way it should be done.

We could solve this in MDIS Wizard by redirecting stderr to /dev/null, but this would work for all static libraries used by MDIS Wizard such as Xlib and thus prevent from seeing probably important messages (see #102).

We could try to use newer fontconfig library and count on the backward compatibility (when running on old distros), but this would also introduce newer libraries that fontconfig depends on and fontconfig dependent. And there is not guarantee, that it would work with old distributions.

dpfeuffer, 2019-04-05:
Because compatibility to older distros is more important than some error/warning messages, I close the issue and add a note to the MDIS under Linux user manual:
Note: If you start mdiswiz from the console, you may see a few "Fontconfig error/warning: ..." messages". You can ignore this messages, which are caused from an incompatibillity between the fontconfig static library linked to mdiswiz and configuration files of your Linux distro.

closed 2019-04-05 by dpfeuffer