MEN-Mikro-Elektronik/13MD05-90 #94: driver revision strings shall indicate submodule revision tags

dpfeuffer, 2019-03-20:
The MDIS revision strings that can be queried with M_getstat(M_MK_BLK_REV_ID) from each MDIS driver shall indicate the GIT revision tags of the low-level driver and kernel lib repos.
The revision strings are currently hard coded in the belonging source files (were stamped from MCVS during check-in).
We have to evaluate how the revision strings could be supported in Git. tbd.

Example:
```
[root@mdis4lin mproj_wizscan]# m_rev_id gpio_3
-----[ MDIS Kernel ]-----
MK - MDIS Kernel: $Id: ident.c,v 2.0 2003/02/21 13:34:33 kp Exp $
-----[ Low Level Driver ]-----
Z17 - Z17 low level driver: $Id: z17_drv.c,v 1.15 2017/05/03 16:52:05 DPfeuffer Exp $ Z34/Z37 model
DESC - Descriptor Decoder: $Id: desc.c,v 1.16 2010/09/02 15:15:08 UFranke Exp $
OSS - MEN Linux Operating System Services: %FSREV LINUX/LIBSRC/oss 2.15 2012-08-07%
-----[ BBIS Kernel ]-----
BK - BDIS Kernel: $Id: bk_module.c,v 1.9 2014/07/29 14:35:51 ts Exp $
-----[ BBIS Driver ]-----
CHAMELEON -   Base Board Handler: $Id: bb_chameleon.c,v 1.48 2013/11/28 17:00:05 ts Exp $
DESC - Descriptor Decoder: $Id: desc.c,v 1.16 2010/09/02 15:15:08 UFranke Exp $
OSS - MEN Linux Operating System Services: %FSREV LINUX/LIBSRC/oss 2.15 2012-08-07%
```

dpfeuffer, 2019-05-06:
After my evaluation how the revision strings could be supported in Git, here are the requirements how to fix it (and how to clean up some outdated MCVS stuff): 

# Overview of MCVS related fields 

| Versions Info                | Location    | Inserted (MCVS)   |  Used for |
| ---------------------------- | ----------- | ----------------- | --------- |
| $Date: 2019/04/29 10:46:01 $ | source file | at file checkin   | -         |
| $Revision: 1.2 $             | source file | at file checkin   | -         |
| $Header: r:/./WINDOWS/TEST/MCVS_FIELDS/mcvs_fields.txt,v 1.4 ... $ | source file | at file checkin | - |
| $Id: mcvs_fields.txt,v 1.1 2019/04/29 13:09:32 DPfeuffer Exp $     | source file | at file checkin | M_getstat(M_MK_BLK_REV_ID) |
| $Log: mcvs_fields.txt,v $ <br> Revision 1.1  2019/04/29 13:09:32  DPfeuffer <br> Initial Revision | source file | at file checkin | - |
| %FSREV WINDOWS/TEST/mcvs_fields 1.1 2019-04-29% | source file | at FILESET checkin | M_getstat(M_MK_BLK_REV_ID) |
| \<date\>2019-04-29\</date\>   | .xml file | at FILESET checkin | mdiswiz package date |
| \<revision\>1.3\</revision\>  | .xml file | at FILESET checkin | mdiswiz package rev |

## Required modifications for clean MCVS to Git migration
The old MCVS fields are no longer updated from Git. The stamped information in source files (*.c,*.h,*.mak,...) are outdated and confusing.

### R1: The following lines with MCVS fields shall be removed in all source files
- $Date
- $Revision
- $Header
- $Id (if not used in variable assignment!)
- $Log (already done)

### R2: The following lines shall be modified in all source files
- The $Id/%FSREV lines in *.c/*.h files shall be changed for Git usage
  Example:
  ```static const char IdentString[]=MENT_XSTR(MAK_REVISION);```
  Notes:
  - This line must be placed somewhere after ```#include <MEN/men_typs.h>``` because of the used MENT_XSTR macro.
  - In pgrogram sources, it must be checked that the revision information will be properly displayed at runtime.
- Some Author lines have two $ signs by mistake. Please remove it.
  Example:
  ```
  $Author: cs $
  ```
  ==>
  ```
  Author: cs
  ```
- The "(MDIS V4.x)" and "MDIS 4.0" like strings in source comments shall be removed:
  Example:
  ```
  Project: M99 module driver (MDIS V4.x)
  ```
  ==>
  ```
  Project: M99 module driver
  ```

See https://github.com/MEN-Mikro-Elektronik/13M099-06/pull/1

### R3: All belonging *.mak files shall be modified to support Git version information

Example:
```
# the next line is updated during the MDIS installation
STAMPED_REVISION=

DEF_REVISION=MAK_REVISION=$(STAMPED_REVISION)
MAK_SWITCH=$(SW_PREFIX)MAC_MEM_MAPPED \
           $(SW_PREFIX)$(DEF_REVISION)
```

See https://github.com/MEN-Mikro-Elektronik/13M099-06/pull/1

### R4: INSTALL.sh shall update the "STAMPED_REVISION=..." line in .mak files
During MDIS installation, INSTALL.sh shall update the "STAMPED_REVISION=..." line of all *.mak files, if this line exists.
The following git commit information of the repo (13MD05-90 or submodule) where the .mak file belongs to shall be assigned to STAMPED_REVISION: 
[... 62 lines cut ...]
13M099-06_02_15-0-g31531d1-dirty_2019-02-21
MBUF - Buffer Manager: $Id: mbuf.c,v 1.18 2010/04/21 16:51:11 CRuff Exp $
DESC - Descriptor Decoder: $Id: desc.c,v 1.16 2010/09/02 15:15:08 UFranke Exp $
OSS - MEN Linux Operating System Services: %FSREV LINUX/LIBSRC/oss 2.15 2012-08-07%
-----[ BBIS Kernel ]-----
BK - BDIS Kernel: $Id: bk_module.c,v 1.9 2014/07/29 14:35:51 ts Exp $
-----[ BBIS Driver ]-----
D203    Base Board Handler: $Id: bb_d203.c,v 1.18 2013/11/28 17:00:24 ts Exp $
DESC - Descriptor Decoder: $Id: desc.c,v 1.16 2010/09/02 15:15:08 UFranke Exp $
OSS - MEN Linux Operating System Services: %FSREV LINUX/LIBSRC/oss 2.15 2012-08-07%
```

### Revision info from tools

```
[root@mdis4lin mproj_m99]# m99_latency
...
13M099-06_02_15-0-g31531d1-dirty_2019-02-21
(c) 2003..2010 by MEN mikro elektronik GmbH
```

### MDIS Wizard revision info about installed packages

mdiswiz: Packages --> Show Installed
13m09906    13M099-06_02_15-0-g31531d1-dirty

dpfeuffer, 2019-05-06:
Because this issue affects nearly all 13MD05-90 submodules and we are currently investigate the MDIS license usage, the issue https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/116 should also be considered during the modifications of the submodules to reduce the workload.

KonradJPE, 2019-05-08:
According to git commands in R4, R5:

```
    <revision>: result of git describe --dirty --long --tags --always
    <date>: result of git --no-pager show -s --date=short --format=format:"%cd%n"
```
We have decided that .git directory will not be provided with release package, so we cannot obtain any data from git directly. https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/61

For release package HISTORY dir is provided. For every submodule below files are available:
```
        git log > 13MD05-90_history.txt"
        git remote -v > 13MD05-90_url.txt"
        git rev-parse --verify HEAD > 13MD05-90_version.txt"
        git describe --exact-match --tags $(git rev-parse --verify HEAD)  &> "/13MD05-90_tag.txt"
```
Questions are:
1. Should we create another file with results of commands that will be added to HISTORY dir? 
```
    <revision>: result of git describe --dirty --long --tags --always
    <date>: result of git --no-pager show -s --date=short --format=format:"%cd%n"

```
2. Or should we change the definition of STAMPED_REVISION
`STAMPED_REVISION="<tag-numberOfCommits-abbrevCommit>_<date>" `
to data that are already available in HISTORY files ? example: 
`STAMPED_REVISION="<commit_id>_<tag - (only if available)>" `

dpfeuffer, 2019-05-08:
I'm confused.
Yes, you are right, we decided to remove .git from the delivery package, but it is still included in the 13MD05-90_02_00 release archive files. See https://github.com/MEN-Mikro-Elektronik/13MD05-90/releases/tag/untagged-0cf3f14fb95f35efe031
The HISTORY dir is present after the MDIS installation under /opt/menlinux.
Therefore, I assumed, that the INSTALL script gets the git information with git from .git folder and I forgot decission #61 .
When I look into the /opt/menlinux/HISTORY/* files, I see that all submodules url and version points to 13MD05-90:
```
dpfeuffer@vm-suse:/opt/menlinux/HISTORY> cat 13M034-06_url.txt 13M066-06_url.txt 13Y004-06_url.txt
origin  https://github.com/MEN-Mikro-Elektronik/13MD05-90 (fetch)
origin  https://github.com/MEN-Mikro-Elektronik/13MD05-90 (push)
origin  https://github.com/MEN-Mikro-Elektronik/13MD05-90 (fetch)
origin  https://github.com/MEN-Mikro-Elektronik/13MD05-90 (push)
origin  https://github.com/MEN-Mikro-Elektronik/13MD05-90 (fetch)
origin  https://github.com/MEN-Mikro-Elektronik/13MD05-90 (push)
dpfeuffer@vm-suse:/opt/menlinux/HISTORY> cat 13M034-06_version.txt 13M066-06_version.txt 13Y004-06_version.txt
966a6ae6a2e7c4047edb861a1cda7e27a5003fba
966a6ae6a2e7c4047edb861a1cda7e27a5003fba
966a6ae6a2e7c4047edb861a1cda7e27a5003fba
```
If we stick to decision #61:
To Q1.
Possible solution, but more files.

To Q2.
I'm missing the date information, that is important for me.

3.
Could we change the _version.txt files from complete commit hash
e.g. ```966a6ae6a2e7c4047edb861a1cda7e27a5003fba```
To the demanded format, \<tag-numberOfCommits-abbrevCommit\>_\<date\>
e.g. ```13m09906 13M099-06_02_15-0-g31531d1```
or are the _version.txt files already evaluated by someone (tool, user)?

I think the new format would be also more useful for the customer. What do you think about it?

KonradJPE, 2019-06-05:
@dpfeuffer I have additional question regarding $Id identifier. Example below.
```
static char* Ident( void )
{
    return( "Z61 - Z61 low-level driver: "
#ifdef MAC_BYTESWAP
    "swapped "
#else
    "non swapped "
#endif
    "$Id: z61_drv.c,v 1.3 2009/07/10 14:12:03 CRuff Exp $" );
}
```

Should i just replace all information with IdentString?
```
static char* Ident( void )
{
    return( (char*) IdentString );
}
```

dpfeuffer, 2019-06-05:
If we have just one or a few driver which provides the information of swapped or non swapped, then I would say we can skip this information.
If most drivers with swapped variants provide this distinct revision info, then we should solve it like this:

```
#ifdef MAC_BYTESWAP
    static const char IdentString[]=MENT_XSTR_SFX(MAK_REVISION, " (swapped)");
#else
    static const char IdentString[]=MENT_XSTR_SFX(MAK_REVISION, " (non swapped)");
#endif
```
SFX stands for suffix and the MENT_XSTR_SFX macro (tbd.) shall add the second argument to the MAK_REVISION.
Example: ```"13M099-06_02_15-0-g31531d1-dirty_2019-02-21 (swapped)"``` 
MENT_XSTR_SFX macro shall be defined in men_types.h

KonradJPE, 2019-06-06:
> If we have just one or a few driver which provides the information of swapped or non swapped, then I would say we can skip this information.

There are 3 drivers that provides this swapped/non swapped information, so i suppose we can skip this. 

Should I also omit information about model? Please find example below - Only several modules use this information. 

```
static char* Ident(void)
{
	return ("Z17 - Z17 low level driver: $Id: z17_drv.c,v 1.15 2017/05/03 16:52:05 DPfeuffer Exp $"
			#ifdef Z17_MODEL_Z127
				" Z127 model"
			#else
				" Z34/Z37 model"
			#endif
			#ifdef Z127_NOIRQ
				" - no IRQs"
			#endif
			);
}
```

dpfeuffer, 2019-06-06:
This model info may be important to know for our support team, when a customer run into problems. It's the same kind of additional information as with the swap/non-swap info. Because we may have in sum up to ten drivers with additional information, please try to use a new macro (MENT_XSTR_SFX) to add this info in brackets as I suggested.
For the z17 driver we have then four cases:
- " (Z127 model)"
- " (Z127 model, no IRQ)"
- " (Z34/Z37 model)"
- " (Z34/Z37 model, no IRQ)"

KonradJPE, 2019-06-11:
Requirements R1-R7 should be now met.
Due to a large number of minor changes that have been done automatically by script, some tests should be performed.
 
After performing compilation test, and performing some simple tests (for R1-R7) requirements I will close the issue.

mlassota, 2019-06-18:
Pull request https://github.com/MEN-Mikro-Elektronik/13MD05-90/pull/115 allowed all LL modules to display version info, but there are couple of modules that do not include this information. Would you like us to do the same for other modules?

dpfeuffer, 2019-06-24:
Yes, see
>R6: The revision info of all used SW modules shall be updated

above.

mlassota, 2019-07-02:
All requirements seem to be met.

closed 2019-07-02 by mlassota