MEN-Mikro-Elektronik/13MD05-90 #95: The generated documentation of all submodules shall be up to date

dpfeuffer, 2019-03-22:
We will add a new requirement like:
<MDIS_ConfRule_XXXX> SW Module Documentation
The documentation generated from source files shall be updated (e.g. with doxygen) after changes at sources were made which affects the generated documentation.

I'm afraid that in the past the documentation was often not updated by us after changes to the sources. For the next MDIS release, the documentation of all submodules should therefore be checked to see if it is up-to-date (or simply updated).

KonradJPE, 2019-05-06:
New tools have been added into modules. Check if tools are available in doxygen documentation.
13Z015-06
13Z005-06
13Y001-06
13M075-06
13M062-06
13M054-06
13M022-06
13AD78-06

mlassota, 2019-05-29:
I've updated Doxygen generated documentation for all modules having in-code Doxygen comments.

Also I've added MDIS_COMMIT tag which is always commented out and has a commit hash of the commit the documentation was generated for. It's only for the purpose of automated documentation generation.

closed 2019-05-29 by mlassota