MEN-Mikro-Elektronik/13MD05-90 #96: WINDOWS directory with mdiswiz.exe.

KonradJPE, 2019-03-22:
Should WINDOWS directory be removed from MDIS repo (13MD05-90/MDISForLinux/WINDOWS)? 
It contains mdiswiz executable for Windows.

dpfeuffer, 2019-03-22:
I'm not able to decide it. I will discuss it with Florian and Thomas Schnürer today as soon as possible. Then I will give you feedback.

dpfeuffer, 2019-03-22:
Thomas told me, that mdiswiz.exe makes only sense for ElinOs Windows build hosts and it was used in the past only from one customer. And we will skip ElinOs support, as far as I know.
So, please remove it!
Your question comes just in time because I'm rewriting the MDIS User Manual for Linux. :-) I will remove the description of the existence of the Windows folder in the User Manual now.

KonradJPE, 2019-03-22:
Thanks for your answer. This directory will be removed from the next release - 13MD05-90_02_01.

dpfeuffer, 2019-03-22:
Is it possible, to remove it from 13MD05-90_02_00_b2?

dpfeuffer, 2019-03-22:
Ok, my request comes too late ;-) No problem, you can do this for the next release 13MD05-90_02_01.

closed 2019-03-22 by KonradJPE