MEN-Mikro-Elektronik/13MD05-90 #102: mdiswizard shall be able run through ssh with X-forwarding

FlorianRauh, 2019-04-04:
A lot of customers are trying a to build a tiny Linux image without GUI (X-Window support  like LXDE,GNOME,KDE)
therefore it will help our customers a lot if the mdiswiz is able to run  through ssh.
error messages:
```

> men@men-MEN-G22-F22P-CB70-A22-SC:~/work/dummymdis$ sudo /opt/menlinux/BIN/mdiswiz
> Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
> Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
> Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
> mdiswiz: cannot connect to X server localhost:10.0
men@men-MEN-G22-F22P-CB70-A22-SC:~/work/dummymdis$

```

dpfeuffer, 2019-04-05:
I used mdiswiz quite often remotely via ssh X forwarding without any problems (from OpenSUSE host or Windows with Mobaxterm to CentOS and Ubuntu targets).
Maybe it's just a configuration problem. You have to enable "ForwardX11 yes" in .ssh/config or specify ssh -X.

mlassota, 2019-04-05:
Well, I had this issue before too, but I solved it by replacing the host part ("localhost") in the DISPLAY environment variable (on OS where mdiswiz is located) with corresponding IP, e.g.:
$ echo DISPLAY
localhost:10.0
$ export DISPLAY=127.0.0.1:10.0

But maybe there's something I can do to mdiswiz so it had no problem with X server connection. Will see.

FlorianRauh, 2019-04-05:
Host mobaxterm V11.0 or V11.1 on Windows 7 - Target Lubuntu 18.04 (LXDE)
[image]

the workaround doesn´t help.

```
root@men-MEN-G22-F22P-CB70-A22-SC:/home/men# export DISPLAY=127.0.0.1:10.0
root@men-MEN-G22-F22P-CB70-A22-SC:/home/men# /opt/menlinux/BIN/mdiswiz
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
MobaXterm X11 proxy: Unsupported authorisation protocol
mdiswiz: cannot connect to X server 127.0.0.1:10.0
root@men-MEN-G22-F22P-CB70-A22-SC:/home/men# uname -a
Linux men-MEN-G22-F22P-CB70-A22-SC 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
root@men-MEN-G22-F22P-CB70-A22-SC:/home/men#
```

mlassota, 2019-04-05:
> MobaXterm X11 proxy: Unsupported authorisation protocol
> mdiswiz: cannot connect to X server 127.0.0.1:10.0

That's because you run mdiswiz on "root" account.
Try to export DISPLAY variable as before but when on regular user account, e.g. "men". Then run mdiswiz on "men" account using sudo.
E.g.:
`men@men-MEN-G22-F22P-CB70-A22-SC:~$ export DISPLAY=127.0.0.1:10.0`
`men@men-MEN-G22-F22P-CB70-A22-SC:~$ sudo /opt/menlinux/BIN/mdiswiz`
At least it works for me.
I'm trying to figure out a solution for this.

FlorianRauh, 2019-04-08:
Dear Marcin,
you are right. I agree. Please try to find a solution as I don´t need the workaround e.g. with "gedit"

mlassota, 2019-04-09:
### Explanation
MDIS Wizard is 32-bit x86 architecture statically linked Linux application.
However there's a part of glibc library (that MDIS Wizard is statically linked with) that uses dynamic loading of libraries. And it uses it to load library (e.g. libnss_myhostname.so) that resolves host names which in our case is "localhost".
So Linux tries to find and load such a library, but since MDIS Wizard is 32-bit it has to find and load 32-bit library. On 64-bit distros you have no 32-bit libraries installed by default thus it cannot find and load and it cannot resolve host name (that come from environment variable DISPLAY), so that is why you get the "mdiswiz: cannot connect to X server localhost:10.0" error.
If you would run MDIS Wizard on 32-bit distribution or if you would run 64-bit MDIS Wizard on 64-bit distribution there would be no problem.

### Solution
- Simple but user-side:
On 64-bit distributions user can install 32-bit libnss-myhostname dynamic library:
```sudo apt install libnss-myhostname:i386```

- Complex but MDIS Wizard side:
Make glibc static library that uses different approach when it comes to host names resolving (*--enable-static-nss* compile option), and then link it with MDIS Wizard. This will probably increase mdiswiz size a bit.

dpfeuffer, 2019-04-09:
Have you tried this:
Add "X11UseLocalhost no" to /etc/ssh/sshd_config and restart the SSH server.

mlassota, 2019-04-09:
> Add "X11UseLocalhost no" to /etc/ssh/sshd_config and restart the SSH server.

What I get with above setting is:
```mdiswiz: cannot connect to X server men:10.0```
and it's all because it cannot resolve "men" hostname cause no 32-bit libnss-* can be found.
That is why ```export DISPLAY=127.0.0.1:10.0``` works.

dpfeuffer, 2019-04-09:
I assume you are logged in from host to target with x11 forwarding (ssh -X).

My host configuration:
```
[men@mdis4lin ~]$ cat .ssh/config
Host *
  ForwardAgent yes
  AddKeysToAgent yes
  ForwardX11 yes
```

My active target configuration in /etc/ssh/sshd_config
```
HostKey /etc/ssh/ssh_host_rsa_key
HostKey /etc/ssh/ssh_host_ecdsa_key
HostKey /etc/ssh/ssh_host_ed25519_key
SyslogFacility AUTHPRIV
AuthorizedKeysFile  .ssh/authorized_keys
PasswordAuthentication yes
ChallengeResponseAuthentication no
GSSAPIAuthentication yes
GSSAPICleanupCredentials no
UsePAM yes
X11Forwarding yes
X11UseLocalhost no
AcceptEnv LANG LC_CTYPE LC_NUMERIC LC_TIME LC_COLLATE LC_MONETARY LC_MESSAGES
AcceptEnv LC_PAPER LC_NAME LC_ADDRESS LC_TELEPHONE LC_MEASUREMENT
AcceptEnv LC_IDENTIFICATION LC_ALL LANGUAGE
AcceptEnv XMODIFIERS
Subsystem   sftp    /usr/libexec/openssh/sftp-server
```

awujek, 2019-04-10:
Could you try to execute command "xhost +" on the host system as regular user (not root) and then check once more?

Adam

mlassota, 2019-04-10:
>Could you try to execute command "xhost +" on the host system as regular user (not root) and then check once more?
>
>Adam

It didn't work, Adam.

mlassota, 2019-04-10:
I did set the exact configuration that Dieter has, but it didn't work.

Then I set up CentOS 7 [CentOS Linux release 7.5.1804 (Core)] virtual machine and only set "X11UseLocalhost no" in "/etc/ssh/sshd_config" (other options were set by default). mdiswiz works like a charm.

mlassota, 2019-04-10:
So it turns out that it's the matter of sshd (openssh-server). To be specific it's about the compile option ```--with-ipaddr-display```. On CentOS sshd is compiled with this flag, but on Ubuntu sshd is compiled without this flag.
With this flag sshd exports DISPLAY environment variable with host IP not with host name. So since mdiswiz cannot resolve hostnames it cannot connect to X server.

So for CentOS we can add Dieter's way to list of available solutions.
Nevertheless Florian or Dieter should decide what to do with it.

dpfeuffer, 2019-04-11:
I can confirm your test results with this test:
Host: OpenSuse
Target: Ubuntu 16.04.1 LTS

## user login
```
pfeuffer@SW-53-lubuntu-32:~$ echo $DISPLAY
localhost:10.0
dpfeuffer@SW-53-lubuntu-32:~$ /opt/menlinux/BIN/mdiswiz 
mdiswiz: cannot connect to X server localhost:10.0
dpfeuffer@SW-53-lubuntu-32:~$ DISPLAY=127.0.0.1:10.0
dpfeuffer@SW-53-lubuntu-32:~$ /opt/menlinux/BIN/mdiswiz 
```

## root login
```
root@SW-53-lubuntu-32:~# echo $DISPLAY 
localhost:10.0
root@SW-53-lubuntu-32:~# /opt/menlinux/BIN/mdiswiz 
mdiswiz: cannot connect to X server localhost:10.0
root@SW-53-lubuntu-32:~# DISPLAY=127.0.0.1:10.0
127.0.0.1:10.0
root@SW-53-lubuntu-32:~# /opt/menlinux/BIN/mdiswiz 
```

So under Ubuntu, the workaround is to set "DISPLAY=127.0.0.1:10.0" either as logged in user or root, then call mdiswiz.

@mlassota 
Could you please test your suggestion, if the effort is not so high:
**Complex but MDIS Wizard side:
Make glibc static library that uses different approach when it comes to host names resolving (--enable-static-nss compile option), and then link it with MDIS Wizard. This will probably increase mdiswiz size a bit.**

mlassota, 2019-04-12:
> Could you please test your suggestion, if the effort is not so high:
Complex but MDIS Wizard side:
Make glibc static library that uses different approach when it comes to host names resolving (--enable-static-nss compile option), and then link it with MDIS Wizard. This will probably increase mdiswiz size a bit.

Unfortunately it took me to many time to just compile static glibc (it also required patching glibc), and gcc just does not want to link with this custom glibc, and I think it will take some time to make it work. Therefore maybe one of abovementioned solution could be acceptable.

Another solution that comes to my mind is just to provide 64-bit mdiswiz, so teher would be two (32-bit and 64-bit).

dpfeuffer, 2019-04-12:
I close this issue because a workaround exists to use mdiswiz with x-window forwarding.
I will add the following note to the MDIS User Manual (there was already one in the past, but not in the 13MD05-90_02_00 version):

If the X Server is not running on the same machine as mdiswiz, mdiswiz may have a problem to connect to the remote X-server. Whether the problem occurs or not depends on the shared libraries installed on the host that runs mdiswiz.
Workaround: Replace the host name in the DISPLAY environment variable by the host IP address (e.g localhost:10.0 --> 127.0.0.1:10.0).

closed 2019-04-12 by dpfeuffer