MEN-Mikro-Elektronik/13MD05-90 #103: test fpga_load_x86-32 at x86_64 distros and rename it to fpga_load

dpfeuffer, 2019-04-09:
Please add a test case for all Linux distros for the verification that fpga_load_x86-32 can be executed without errors.
If the test succeeds:
  - please rename it to fpga_load and remove fpga_load_x86-64.
  - I have to change the user manual accordingly.

KonradJPE, 2019-04-09:
As far as I understand, this tests: 

> test case for all Linux distros for the verification that fpga_load_x86-32 can be executed without errors.

is unnecessary. We are planning to compile fpga_load from sources before scanning the system, and no fpga_load binary will be provided to user. What is more, i think that dynamically linked fpga_load binary will be sufficient. 

Please correct me if, i am wrong.

dpfeuffer, 2019-04-09:
The fpga_load (and mm_ident, mdiswiz) binary shall still be included in 13MD05-90. Please see the requirement in issue #84 "MDIS Install/Scan Script Requirements".
We have two reasons for this:
1. Customers who just want to update/flash a new FPGA programming file into the FPGA (without using any MDIS driver in the system) requires the binary and shall take it from the 13MD05-90 package, without the necessary to build it (and have to setup a compiler).
2. It should be possible to copy fpga_load, mm_ident and scan_system.sh from the 13MD05-90 package to a tiny target system (where no compiler is available) and perform a system scan.

Of course, in both cases this works only on x86 architectures, but this is to 90% the customer case.
Compiling fpga_load and mm_ident is optional, e.g. for non x86 architectures or when an up to date tool is required.

KonradJPE, 2019-04-29:
> Please add a test case for all Linux distros for the verification that fpga_load_x86-32 can be executed without errors.

All tests for Release 02_00 have been using fpga_load_x86-32. So test case is unnecessary. 
Binary fpga_load_x86-64 has been removed from repo.

closed 2019-04-29 by KonradJPE