MEN-Mikro-Elektronik/13MD05-90 #104: Fix package description .xml files / advice about missing internal tools

dpfeuffer, 2019-04-12:
I discovered the fact, that some internal tools are missing in MDIS for Linux since the MCVS to Git migration. Note: Internal tools are SW modules with the *internal* attribute, e.g.:
```<swmodule internal="true">```
To examine the 13MD05-90_02_00 delivery for missed internal tools, I wrote a new check_pkgdesc.sh script and added it to https://github.com/MEN-Mikro-Elektronik/mdis_tools_dev.
I got this result from the script:
```Result: 45 .xml files checked, 24 errors in 15 .xml files```
Where
1. for 1 .xml file / lowlevel package I added already a pull request: https://github.com/MEN-Mikro-Elektronik/13M066-06/pull/1
2. in 4 .xml files, just the \<docroot\> entry is wrong
3. in 10 lowlevel packages, internal tools are missing

**To 2. Please fix the \<docroot\> entry in the following .xml files:**
Note: I won't create pull requests for this little changes.
```
--- 13m07790.xml: Linux UART driver package for M45N/M69N/M77 ---
*** <docroot>=DRIVERS/M077/INSTALL not found
--> must be DRIVERS/M077/DOC/html/index.html

--- 13pp0406.xml: MDIS5 driver package for PP4 MVB PCI104 module ---
*** <docroot>=DRIVERS/MDIS_LL/PP04/DOC/main.html not found
--> must be DRIVERS/MDIS_LL/PP04/DOC/html/index.html

--- 13z00190.xml: Linux driver package for MEN 16Z001 SMB Controller ---
*** <docroot>=DRIVERS/SMB_Z001/DOC/html/index.html not found
--> must be DRIVERS/Z001_SMB/DOC/html/index.html

--- 13z02406.xml: MDIS5 driver package for Z24 SRAM - MMODPROG ---
*** <docroot>=DRIVERS/MDIS_LL/MMODPRG/DOC/html/index.html not found
--> <docroot> line must be removed
```

**To 3: Please proceed the pull requests of the affected submodules**
Note: I will add it in the next two weeks pull requests with the missing internal tools and a few .xml file modifications.

KonradJPE, 2019-04-15:
> 1. for 1 .xml file / lowlevel package I added already a pull request: MEN-Mikro-Elektronik/13M066-06#1

Merged

> To 2. Please fix the <docroot> entry in the following .xml files:

Fixed

KonradJPE, 2019-05-06:
After merging sources, result is as below: 
```
Result: 46 .xml files checked, 0 errors in 0 .xml files
--- SUCCESS --
```

closed 2019-05-06 by KonradJPE