MEN-Mikro-Elektronik/13MD05-90 #105: Commit all changes of the following pay submodules that I will remove

dpfeuffer, 2019-04-25:
We discovered that the following pay MDIS low level driver packages are at GitHub by mistake:
- https://github.com/MEN-Mikro-Elektronik/13M057-06
- https://github.com/MEN-Mikro-Elektronik/13M065-06
- https://github.com/MEN-Mikro-Elektronik/13M079-06
- https://github.com/MEN-Mikro-Elektronik/13Z016-06

This submodules must not be public available due to license issues.
I will move this submodules to a MEN local GIT CVS (Bitbucket).
Please check, if you have any outstanding local commits and merge all commits to the master branches of the submodules.
Please remove also the 13M065-06 submodule reference from 13MD05-90 repo and verify if you are using the 13M065-06 for test purposes. In this case, you have to tbd.

I will also remove the 13M065-06 submodule from the 13md05-90.zip that we provide our customers at www.men.de without changing the 13MD05-90 revision (still 02.00). 
The 13M065-06 content is also included in all the delivery packages at https://github.com/MEN-Mikro-Elektronik/13MD05-90/releases. I'm not yet sure how to deal with this.

KonradJPE, 2019-04-25:
Hello Dieter,
I have merged all changes into master for submodule 13M065-06. 
Module is now removed from 13MD05-90 repo.

mlassota, 2019-04-25:
I've removed 13M065-06 from master branch.

dpfeuffer, 2019-04-26:
1) I moved the four pay MDIS low level driver packages from GitHub to MEN local GIT CVS (Bitbucket).

2) I found a solution to hide the 13MD05-90_*.zip/.tar.gz release packages by saving each release as draft. Draft releases won’t be seen by the public unless it’s published. My suggestion is to store all releases as draft. Customers shall visit www.men.de for official releases.

3) I also modified the README.md file directly at the master branch to add a link to the MEN download site to redirect users to the official MDIS release. (Sorry for that quick hack, normally only JPE should commit to master.)

closed 2019-04-26 by dpfeuffer