MEN-Mikro-Elektronik/13MD05-90 #106: mdiswiz: internal tools are selected when ll pkg was installed from mdiswiz

dpfeuffer, 2019-04-26:
13MD05-90_02_00
openSUSE Leap 15.0

When a low level driver package with internal tools is part of 13MD05-90, the internal tools are listed unselected under Sowftware Modules in mdiswiz. This is the demanded behavior (ok).
When I install a low level driver package from mdiswiz (Packages->Install Driver) AND reopen the MDIS project, then all internal tools are selected. This is not ok.

My test was:
1. Add F18 CPU
2. Add F205 M-Module Carrier
3. Add M66 M-Module
4. Show Software Modules: m66_irq is not selected (ok)
5. Create ll package zip file from https://github.com/MEN-Mikro-Elektronik/13M066-06.git: ```zip -r 13m06606.zip DRIVERS/ INCLUDE/ PACKAGE_DESC/ README.md```
6. Install 13m06606.zip from mdiswiz (Packages->Install Driver)
7. Show Software Modules: m66_irq is not selected (ok)
8. Save MDIS project, reopen mdiswiz with saved project
9. Show Software Modules: m66_irq is selected (err)
10. Create new MDIS project like steps 1..3
11. Show Software Modules: m66_irq is selected (err)

dpfeuffer, 2019-04-26:
I discovered now the cause of the problem:
Obviously, mdiswiz does not consider the internal tag at all. The internal tools were just not selected because the sources of the internal tools were not included.
So my assumption about the handling of internal tools from mdiswiz was wrong :-(
I will change the issue label from bug to enhancement, but we definitely need this behavior:
All software modules with the "internal" attribute in the .xml shall not be build from mdiswiz per default.
This behavior is already described in the MDIS user manual.

KonradJPE, 2019-05-17:
> All software modules with the "internal" attribute in the .xml shall not be build from mdiswiz per default.
> This behavior is already described in the MDIS user manual.

Shall we also add this behaviour for scan_system.sh script?
This could be done by adding question before scanning the hardware. 
"Include internal tools in Makefile (y/n) ?"

dpfeuffer, 2019-05-17:
No, the Makefile created from scan_system.sh shall not contain any internal tools. If the user want's to use internal tools, he could open the scan_system.sh created MDIS project with mdiswiz and then add the demanded internal tools.

KonradJPE, 2019-05-20:
> All software modules with the "internal" attribute in the .xml shall not be build from mdiswiz per default.
> This behavior is already described in the MDIS user manual.

Solved. Now user has to manually select 'internal' tools for build. 

It is worth noting (maybe in user manual) that after opening project and saving it again 'internal' modules will be automatically removed from makefile. This is due the fact, that mdiswiz does not read Makefile at all, it is creating it from scratch every time. 

@dpfeuffer Is it ok for you? If yes, then please close the issue.

dpfeuffer, 2019-05-21:
This behavior is not ok, because it is really error-prone for the customer. The problem is not only restricted to "internal" tools. I'm afraid, that we have a general design problem how system scan and mdiswiz works together. This behavior may also affect issue #119.
I close this issue and summarize the problems in #121 .

closed 2019-05-21 by dpfeuffer

KonradJPE, 2019-06-19:
> It is worth noting (maybe in user manual) that after opening project and saving it again 'internal' modules will be automatically removed from makefile. 

Fixed. If internal module exists in makefile, then mdiswiz treats this sw module as non internal and it is not removed.