MEN-Mikro-Elektronik/13MD05-90 #108: Add support for GROUP_ID_x identifier in scan_system.sh script.

KonradJPE, 2019-04-29:
Are the groups of IP cores still used in chameleon boards (GROUP_ID_x identifier)? 
For now it seems that scan_system.sh script does not support it, but mdiswiz does.
 

> Yes, it is still used and must be supported. However it is rarely used. E.g. the QSPI driver uses this feature for certain driver variants.

Additional information in document: 
[GROUP_ID_x.pdf](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3127159/GROUP_ID_x.pdf)

KonradJPE, 2019-06-19:
@dpfeuffer , @JSierl 
Is it possible to send us hardware that is using GROUP_ID identifier, so we could test and implement this feature into scan_system script?

JSierl, 2019-06-24:
Please postpone this at the moment.

dpfeuffer, 2019-07-26:
Just a remark: The 13Z062-06 DMA test driver also uses groups, but this driver is currently only available in MEN MCVS.

dpfeuffer, 2020-02-14:
I close this issue, because GROUP_ID is currently not used from any MDIS driver of 13MD05-90_02_01.

closed 2020-02-14 by dpfeuffer