MEN-Mikro-Elektronik/13MD05-90 #112: mdiswiz: Altering Compact PCI Slot number has no effect

dpfeuffer, 2019-05-03:
mdiswiz 3.00.030-linux-14.0

Example configuration:
CPU: F18
CPCI Bus Component: F205
Altering the slot number (2, 3, ...) has no influence to the generated system.dsc file.

I'm sure, that in the past the chosen slot number changed die location parameters in the system.dsc file (otherwise, the parameter would be useless).

Please investigate this problem.

dpfeuffer, 2019-05-09:
I suggest, that for PCI busses, the default shall be to enter the physical slot number OR optional (in case the physical slot number doesn't work) the PCI bus and device number.
For PCI serial busses, the PCI bus number shall be entered (PCI device number is always 0).
I think this was the behavior in the past (except the improvement to enter optional the PCI bus and device number for PCI busses).
Please check the commits when/why it was this changed from Thomas Schnürer. He told me, that it was changed because some assignment problems between physical slot and PCI bus/dev number. But it works for me in the past.
We cannot keep the current behavior.

KonradJPE, 2019-06-24:
Behaviour described in this issue is caused by the fact, that we have removed this parameter "PCI_BUS_SLOT" from system.dsc - this was due to inconsistency between system.dsc generated by scan_system.sh and after saving the same project in mdiswiz.
Mdiswiz was always adding PCI_BUS_SLOT but removing PCI_BUS_NUMBER & PCI_DEVICE_NUMBER. https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/62

Now to solve current issue we would like to make some adjustments to above behaviour.
There is no point having both PCI_BUS_SLOT and PCI_BUS_NUMBER & PCI_DEVICE_NUMBER simultaneously because it would confuse user. (From our knowledge PCI_BUS_NUMBER & PCI_DEVICE_NUMBER takes precedence over PCI_BUS_SLOT in MDIS kernel.)

**Please check proposed solution:**
When mdiswiz detects that system.dsc already includes PCI_BUS_NUMBER & PCI_DEVICE_NUMBER for CompactPCI, then field to change PCI_BUS_SLOT will be greyed out/invisible.
To edit this field , user will have to select check box with question "Use PCI_BUS_SLOT instead of  PCI_BUS_NUMBER and PCI_DEVICE_NUMBER ?".
When the project is created from scratch then PCI_BUS_SLOT might be used by default in Wizard.
Additionally we will also add corresponding edit fields to the dialog window for PCI_BUS_NUMBER & PCI_DEVICE_NUMBER when the check box will not be selected.
This should keep consistency between mdiswiz/scan_system.sh.

If you have any doubts/concerns please let us now, so we will find the best solution.

dpfeuffer, 2019-06-25:
Your proposed solution sounds good to me. 
>(From our knowledge PCI_BUS_NUMBER & PCI_DEVICE_NUMBER takes precedence over PCI_BUS_SLOT in MDIS kernel.)

That's true, but it's the BBIS driver (PCI BBIS, Chameleon BBIS) that deals with the PCI parameters and provides these precedence.

KonradJPE, 2019-06-26:
> Please check the commits when/why it was this changed from Thomas Schnürer. He told me, that it was changed because some assignment problems between physical slot and PCI bus/dev number. But it works for me in the past.

Could you please provide somehow history of commits in mdiswiz repository or provide us the whole repository before it was released in github?

dpfeuffer, 2019-06-27:
The problem is, that the MDIS Wizard was maintained in MCVS before the move to GitHub and the migration from MCVS to Git was made by one initial git commit instead of committing all available MCVS versions as separate commit. Therefore, it isn't possible to get older versions from the git repo.
I will send you an email with two mdiswiz versions (2.20 and 2.21) taken from MCVS. According the MCVS log, the PCI bus/dev modification should had happened between them.

KonradJPE, 2019-07-05:
> For PCI serial busses, the PCI bus number shall be entered (PCI device number is always 0).

Should we remove the possibility to use BUS_PATH and BUS_SLOT for PCI serial devices in mdiswiz?
In this way, user will have to manually set proper BUS number (or it will be read from system.dsc).

dpfeuffer, 2019-07-08:
mdiswiz 3.00.030-linux-14.0
I hoped, that PCIe support is proper implemented. However, I now realized, that if I use a F23P as CPU type and add manually a MEZZ_CHAM instance to CompactPci bus and one to CompactPci Serial slot x, for both instances I get
```
PCI_BUS_NUMBER = U_INT32 0x0
PCI_DEVICE_NUMBER = U_INT32 0x0
```
For the cPCISer instance, I'm not able to specify the PCI bus number and for cPCI instance I can choose only the physical slot. So there is no possibility to configure the instances properly from MDISWIZ.
The goal should be to implement the right cPCI and cPCISer parameters for the F23P in MDISWIZ and to have the option to overwrite the hard coded values by specifying bus number and device number (cPCI only). In the past, the cPCI configuration of MDISWIZ was suitable, but I have to check, if it is possible to provide a proper cPCISer configuration as well. It depends on the used backplane and I'm not sure, if one config would work for all backplanes.

By the way, I just realized, that the configuration generated from system_scan (13MD05-90_02_00_b2) shows me both mezz_cham instances at location "CompactPCI Serial Slot x". But my testsystem is equipped with one F215 and one G215. The F215 resides at the cPCI bus but mdiwiz displays cPCISerial.   

To your question: Please do not remove BUS_PATH and BUS_SLOT for PCI serial devices in mdiswiz yet. I see, that you did some commits for mdiswiz. Could you please do some configuration testing with your current mdiswiz version?

KonradJPE, 2019-07-08:
> I see, that you did some commits for mdiswiz. Could you please do some configuration testing with your current mdiswiz version?

I will perform tests, and I will add proper configuration fields. I will send you the mdiswiz binary in the coming days for tests, so we will be able to discuss changes.

KonradJPE, 2019-07-10:
I have added below functionalities: 
CPci: 
 

- bus number and device number can be specified by user as alternative for bus slot,

CPci Serial: 

- bus number have to be specified by user.

@dpfeuffer  Could you please take a look into changes and give some feedback? Please find attached mdiswiz binary [mdiswiz.zip](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3377225/mdiswiz.zip) 
If mdiswiz requires some other changes in this field please let us know. 

> By the way, I just realized, that the configuration generated from system_scan (13MD05-90_02_00_b2) shows me both mezz_cham instances at location "CompactPCI Serial Slot x". But my testsystem is equipped with one F215 and one G215. The F215 resides at the cPCI bus but mdiwiz displays cPCISerial.

I could not replicate this with the most current version.

dpfeuffer, 2019-07-11:
I tested the provided mdiswiz for F23P CPU with MEZZ_CHAM instances at cPCI and cPCISerial. The cPCI setting dialog looks good!
The cPCISerial dialog is ok, if it is not possible to define a CPU specific slot/bus setting. I still have to check this:
> In the past, the cPCI configuration of MDISWIZ was suitable, but I have to check, if it is possible to provide a proper cPCISer configuration as well. It depends on the used backplane and I'm not sure, if one config would work for all backplanes.

I detected this problem, when I attach a G204 to a cPCISerial slot:

Specified setting:
[image]

Resulting system.dsc:
```
d203_4 {
    _WIZ_MODEL = STRING G204
    PCI_BUS_PATH = BINARY 0x3c
    PCI_DEVICE_ID = U_INT32 0x0
```

The chosen bus number is ignored and bus path is used. When I attach the G204 to another slot, the bus path is different,
" PCI_BUS_PATH is an array of device IDs of PCI-to-PCI bridges starting from PCI bus 0."
Because the device number is always 0 at cPCISerial the bus path cannot be used at all but maybe a slotNbr-busNbr table (tbd.).

>I could not replicate this with the most current version.

I'll check it with the provided mdiswiz binary tomorrow.

dpfeuffer, 2019-07-12:
>By the way, I just realized, that the configuration generated from system_scan (13MD05-90_02_00_b2) shows me both mezz_cham instances at location "CompactPCI Serial Slot x". But my testsystem is equipped with one F215 and one G215. The F215 resides at the cPCI bus but mdiwiz displays cPCISerial.

I cannot reproduce this with new MDIS installation based on
4aa08a0 19-07-11 15:43 KonradJPE     (HEAD -> jpe-dev, origin/jpe-dev)
I think you fixed this somehow in scan_system.sh.

KonradJPE, 2019-07-12:
> " PCI_BUS_PATH is an array of device IDs of PCI-to-PCI bridges starting from PCI bus 0."
> [2019-07-12 updated]
> For PCIe endpoint devices, the device number is always 0. However, virtual PCI2PCI bridges within the PCIe root complex (or a PCIe switch) resides on the same bus and are distinguished by their individual device number. Therefore the PCI_BUS_PATH can also be used to specify indirect the bus number where a PCIe endpoint device resides.

So should I add checkbox for cPCI-Serial like it is done for cPCI devices? 
Question will look like this: "Use BUS number instead of PCI_BUS_PATH?" 
In contrast to cPCI slot number will be fixed.
After saving, system.dsc should look like this:
```
Checbox: is set
    PCI_BUS_NUM = BINARY xxxx
    PCI_DEVICE_ID = U_INT32 0x0
```
```
Checbox: is unset
    PCI_BUS_PATH = BINARY xxxx
    PCI_DEVICE_ID = U_INT32 0x0
```

PCI_BUS_PATH will be computed by mdiswiz.

dpfeuffer, 2019-07-12:
If we are able to provide a CPU specific PCI_BUS_PATH configuration that works with different backplanes: yes. I'm investigating it at the moment. So please wait...

Please make the following cosmetic changes to get consistent dialogs:
**a) for CompactPCI:**
[image]
Change it to:
"Use PCI Bus and Device Number instead of Slot specific PCI parameters (PCI_BUS_PATH, PCI_BUS_SLOT)" (no ? at the end)
"Determine the actual parameters of the PCI device (e.g. by using lspci command under Linux) and enter it."
red line: "No validity check will be done!"
"PCI Bus Number (PCI_BUS_NUMBER):"
"PCI Device Number (PCI_DEVICE_NUMBER):"
**b) for CompactPCI Serial: (still tbd.)**
"Use PCI Bus Number instead of Slot specific PCI parameters (PCI_BUS_PATH)"
Other lines similar to a)

dpfeuffer, 2019-07-12:
We made some tests with MEN CPU boards F22, F23, F26 and G22 with the following result:
F23:
A plugged G215 board appears independent of the used slot behind a virtual bridge at 1c.0. If an additional G204 was plugged, then the G215 appears behind a virtual bridge at 1c.3.
It is not possible to use PCI_BUS_PATH for F23 CPU.
G22:
PCI_BUS_PATH could be used, but mdiswiz provides currently none.
F26L:
PCI_BUS_PATH could be used, but mdiswiz contains currently wrong parameters.
F22:
PCI_BUS_PATH could be used.

Please skip (comment out) all PCI_BUS_PATH config of PCI Serial slots (not PCI slots!) of all CPUs, except: F19P, F21P, F22P. For these three CPUs, provide the checkbox to optional specify PCI Bus Number (see b) above).
For all other PCI Serial slots without PCI_BUS_PATH, provide the dialog to specify PCI Bus Number directly (not optional, no checkbox).
In both cases, the dialog shall similar to:
[image]
With greyed PCI Device Number, but shall be modified to the new text according a) and b) above.
This decision gives us the possibility to add more correct PCI_BUS_PATH configuration in the future for PCI Serial slots of CPUs.

KonradJPE, 2019-07-19:
I have added functionalities mentioned in this issue. 

Could you please take a look into changes?
If there is nothing more to add regarding this issue, then I would like to close it.

Please find attached binary - [mdiswiz.zip](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3410825/mdiswiz.zip)

dpfeuffer, 2019-07-22:
I made some tests with the new mdiswiz.zip.
A) F23P & MEZZ_CHAM at CPCI:
1. red line: "No validity check will be done!" : Please add the "!"

B) F23P & MEZZ_CHAM at CPCI Serial:
No checkbox displayed because of removed bus path, ok

C) F23P & G204 at CPCI Serial:
Now correct PCI_BUS_NUMBER added, problem fixed,

D) F19P, F21P, F22P & MEZZ_CHAM at CPCI:
Checkbox displayed because of available bus path, ok

E) F19P, F21P, F22P & D203/G204 at CPCI/Serial:
```
HW_TYPE = STRING D203
# ------------------------------------------------------------------------
#        PCI configuration
# ------------------------------------------------------------------------

# ------------------------------------------------------------------------
#        device parameters
# ------------------------------------------------------------------------
# --- C-PCI parameters
PCI_BUS_PATH = BINARY 0x1e
PCI_BUS_SLOT = U_INT32 0x2
PCI_DEVICE_ID = U_INT32 0xf

_WIZ_MODEL = STRING G204
PCI_BUS_PATH = BINARY 0x5c
PCI_DEVICE_ID = U_INT32 0x0
```
2. If possible without great effort: PCI_XXX shall be under PCI configuration section and "C-PCI parameters" shall be deleted. "device parameters" required if e.g. trigger settings are made.
But because of just comments, it's not so important.

3. Another problem, that I also have with the first mdiswiz.zip: When I close the mdiswiz, I get:
ouble free or corruption (out)
Aborted (core dumped)

Tested with:
NAME="openSUSE Leap"
VERSION="15.0"
Linux vm-suse 4.12.14-lp150.12.64-default #1 SMP Mon Jun 17 16:53:50 UTC 2019 (3edfd41) x86_64 x86_64 x86_64 GNU/Linux

NAME="CentOS Linux"
VERSION="7 (Core)"
Linux mdis4lin 3.10.0-862.el7.x86_64 #1 SMP Fri Apr 20 16:44:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

KonradJPE, 2019-07-23:
1. Fixed,
2. Fixed, Please check system.dsc after changes: [system.dsc.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3421240/system.dsc.txt)  - F22 + D203 + G204.
3. We know about this. This problem should not appear in binary included in 13MD05-90 repository.

KonradJPE, 2019-07-23:
It seems to me that this issue can be closed now. Do you agree? 
In case of any doubts please check new mdiswiz binary from 13MD05-90 jpe-dev repo.

dpfeuffer, 2019-07-24:
I did a quick test with mdiswiz (jpe-dev 2b657a2510a48cd70712dad998b201b76fd2d409) and problems 1..3 seems to be fixed now. Thanks!

closed 2019-07-24 by dpfeuffer