MEN-Mikro-Elektronik/13MD05-90 #113: "mcb" and "mcb_pci" to be asked for blacklisted in scan_system.sh

FlorianRauh, 2019-05-03:
mcb and mcb_pci are mainline and are delivered in most standard distributions
mcb and men_lx_chameleon may correlate with each other which leads to kernel panics if loaded at the same time.
Therefore I like to have a warning in scan_system.sh when a FPGA is detected.
"Starting from Linux kernel 3.15, modules mcb and mcb_pci must be
blacklisted in order to avoid inconveniences with the men_lx_chameleon
driver. Have you read the details in User Manual 21MD05-90 pdf ? (y/n)"

KonradJPE, 2019-05-16:
> "Starting from Linux kernel 3.15, modules mcb and mcb_pci must be
> blacklisted in order to avoid inconveniences with the men_lx_chameleon
> driver. Have you read the details in User Manual 21MD05-90 pdf ? (y/n)"

This warning will be printed after scanning, when creation of Makefile/system.dsc file is almost done. 
What should happen when the user choose (no) option? 
Should the Makefile/system.dsc be deleted  and message like below printed? 
"Please blacklist mcb and mcb_pci and then run scan once more."

dpfeuffer, 2019-05-16:
I suggest this warning message:
**Starting with Linux kernel 3.15, the modules mcb and mcb_pci must be blacklisted to avoid problems which could led to a kernel panic. Please refer to the MDIS User Manual 21md05-90.pdf for details.**
If this is the last message from scan_system.sh and Makefile/system.dsc is already created, then I would not ask the user any question
Otherwise, I would add this question:
**Would you like to proceed? (y/n):**
And abort scan_system.sh if the user press n.

@FlorianRauh : What is your opinion?

FlorianRauh, 2019-05-23:
OK Dieter I agree.

But I don´t like the word "problems" . So please:
**Starting with Linux kernel 3.15, the modules mcb and mcb_pci must be blacklisted to avoid inconveniences which could led to a kernel panic. Please refer to the MDIS User Manual 21md05-90.pdf for details.**

KonradJPE, 2019-05-24:
To sum up: 

> Would you like to proceed? (y/n):

When the user choose n (no) or q (quit) option, Makefile is not generated.

dpfeuffer, 2019-12-06:
After some testing, I would like to have the following changes:
1) Detect, if mcb/mcb_pci is available at the system (e.g. by calling modinfo, exit code is only 0 if mcb/mcb_pci is available).
2) If mcb/mcb_pci is available:
- Print this warning message:
**The kernel modules mcb/mcb_pci are available at your system and must be blacklisted to avoid inconveniences which could led to a kernel panic. Please refer to the MDIS User Manual 21md05-90.pdf for details.**
- Without any "Would you like to proceed? (y/n):" question.
3) If mcb/mcb_pci is NOT available:
Print no mcb message at all.

dpfeuffer, 2019-12-17:
When scan is invoked from mdiswiz, I get the mcb related "Would you like to proceed? (y/n):" question.
If this question is removed, there is no longer the necessity to provide the Send box/button in mdiswiz. Or are there any possibilities for other questions from scan_system.sh?
If not: Please remove the Send box/button in mdiswiz.

KonradJPE, 2019-12-30:
> Or are there any possibilities for other questions from scan_system.sh?

Yes, send box/button was added to be able to choose proper driver during scan.
In some cases there are more than one driver match for a HW device (like for 16Z127 IP).

For more details please check issue: 
https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/126

dpfeuffer, 2020-01-07:
Yeah, you're right, I forgot. ;-)

closed 2020-01-07 by dpfeuffer