MEN-Mikro-Elektronik/13MD05-90 #116: MDIS license clearing

dpfeuffer, 2019-05-06:
tbd.

dpfeuffer, 2019-05-13:
R1: Remove all .dsc files from all MDIS repos, because these files are outdated and contain no proper license information.

R2: Change license note from GPL to LGPL for all user mode libs that can be linked from applications. For details, please refer to the upcoming *13MD05-90_SwArchSpec* document.

R3: License and copyright notes shall be added to all files that are required for the build process. For details, please refer to the upcoming *13MD05-90_SwArchSpec* document.

FYI:
To simplify matters, we have decided to reduce the number of MDIS requirements documents. The new  *13MD05-90_SwArchSpec* will replace the following documents:
- InStep/MDIS-_Requirements
- https://github.com/MEN-Mikro-Elektronik/13MD05-90/wiki/Configuration-Management-Plan-for-MDIS
- https://github.com/MEN-Mikro-Elektronik/13MD05-90/wiki/Rules-for-MDIS-low-level-drivers
- https://github.com/MEN-Mikro-Elektronik/13MD05-90/wiki/Milestone-Obligations
The  13MD05-90_SwArchSpec shall be released within the next weeks. We will inform you...

dpfeuffer, 2019-05-20:
R4: Each MDIS repo (13MD05-90 repo and included submodules) shall contain one license file with name LICENSE and the following content:
1)	Repos that contain SW licensed under the GPL only: GPL license text (gpl-2.0.txt)
2)	Repos that contain SW licensed under the LGPL only: LGPL license text (lgpl-2.1.txt)
3)	Repost that contain SW licensed under the GPL as well as SW licensed under the LGPL:
- header with short explanation about GPL and LGPL licensed SW
- GPL license text (content of gpl-2.0.txt)
- LGPL license text (content of lgpl-2.1.txt)
Use this template: https://github.com/MEN-Mikro-Elektronik/13MD05-90/pull/120/files

KonradJPE, 2019-05-27:
> R2: Change license note from GPL to LGPL for all user mode libs that can be linked from applications. For details, please refer to the upcoming 13MD05-90_SwArchSpec document.

Additional questions here: 

a) How to find all affected sources (.c/.h files)?
b) What do you mean by `can be linked from applications`? 
All applications that can be created by user / or applications that are currently available in mdis repository (like hwbug, mm_ident, fpga_load ...)?

I am wondering if below solution for a) is correct. 
1. Find all "_usr" libraries, like
```
chameleon_usr
chameleon_usr_io
chameleon_usr_sw
chameleon_usr_io_sw
oss_usr
dbg_usr
```
2. For all files that are included in makefiles add LGPL licence? 
Example for chameleon_usr this files will be affected: 
```
chameleon_int.h
chameleon.c
men_types.h
chameleon.h
maccess.h
oss.h 
```
I will appreciate your comments :)

dpfeuffer, 2019-05-27:
**To a)**
Now I realize that the separation between GPL and LGPL isn't so easy as assumed. We have some header files (e.g. men_typs.h) which are included from GPL licensed SW modules (e.g. low-level drivers) as well as from LGPL SW modules (user mode libs). I just discussed this topic with other colleagues and our decision is: All sources used for LGPL SW modules shall be licensed under LGPL too. Therefore: All header files included from GPL and LGPL modules shall contain the LGPL note (but no GPL note).
**To b)**
The MDIS comprises libs running in user mode (e.g. MDIS_API, SMB2_API, ..) which are (static or dynamic) linked to our example programs and which are also required for the applications of our customers. Customers do not want to be forced to make their sources public, therefore we have to put all the libs required from applications under LGPL. If the lib sources were licensed under GPL, then the customer would also have to place his derived work (source code) under GPL.
We have different types of user mode libs:
I. General MDIS user mode libs: MDIS_API, USR_OSS, USR_UTL
II. Driver specific API's, which are part of some low-level driver packages: M54_API, MSCAN_API, SJA1000_API, SMB2_API, SMB2_SHC, SMB2_BMC, VME4L_API
III. Some kernel mode libs, that provides a user mode variant: CHAMELEON_USR, OSS_USR, DBG_USR 
**Only I. and II. are necessary for customer applications and shall be under LGPL.**
The special kernel libs with user mode variants (III.) are used from a few MEN tools (fpga_load) but normally not required for custom applications. So this libs shall be licensed under GPL. All our MDIS programs shall be licensed under GPL, so there is no problem with these libs (III.) for us.

I will update the SwArchSpec with this new info.

KonradJPE, 2019-06-11:
Requirements R1-R4 should be met now. 
When the final SwArchSpec document will be released, I will check that there are no new requirements regarding license.
Then I will close this issue.

KonradJPE, 2019-06-11:
@dpfeuffer Additional question regarding MEN Copyrights:
Currently all .c, .h, .mak files supports this style:
```
Copyright (c) <year_of_file_creation>-<year_of_last_file_change>, MEN Mikro Elektronik GmbH
Copyright (c) 2006-2019, MEN Mikro Elektronik GmbH
```
However in SwArchSpec there are copyrights without "(c)": 
`Copyright <year_of_file_creation>-<year_of_last_file_change>, MEN Mikro Elektronik GmbH`

Should we update all sources or leave as it is?

dpfeuffer, 2019-06-24:
You could leave it. However, the copyright info without the additional "(c)" is the better one, because "Copyright" AND "(c)" is redundant (but not wrong).
If additional changes for license/copyright notes may be necessary in future, please do it according the SwArchSpec.

KonradJPE, 2019-07-19:
I have doubts when it comes to license in below src: 
```
MDISforLinux/LIBSRC/MDIS_KERNEL
MDISforLinux/LIBSRC/BBIS_KERNEL
```
What license should be used (LGPL/GPL)?

dpfeuffer, 2019-07-19:
GPL, because NOT directly used (linked) from application SW.
See also MEN_13MD05-90_SA_0190	[DES] Licensing

KonradJPE, 2019-07-23:
MDIS license requirements should be now met.

dpfeuffer, 2019-08-14:
13Z077-90 is a submodule of 13MD05-90
The men_16z077_eth.h file in the current jpe-dev branch contains no licence information as requested with
>R3: License and copyright notes shall be added to all files that are required for the build process. For details, please refer to the upcoming 13MD05-90_SwArchSpec document.

See https://github.com/MEN-Mikro-Elektronik/13Z077-90/blob/jpe-dev/DRIVERS/ETH_16Z077/DRIVER/men_16z077_eth.h

All source files that are required for the build process have to be checked, not only source files that contain already some licence/copyright note.

KonradJPE, 2019-08-21:
@dpfeuffer Could you please check file 13M077-90/DRIVERS/M077/DRIVER/serialP_m77.h ? 
There is Copyright (C) 1997 by Theodore Ts'o.
Should I replace it with the MEN copyrights together with updated GPL note?

dpfeuffer, 2019-08-21:
I know, this is one of the open licence clearing points that I have at my todo list:

1)
There is no information about where this file has been derived from and the GPL version in its header is not specified:
13M077-90/DRIVERS/M077/DRIVER/serialP_m77.h
See also https://www.usenix.org/conference/lisa14/speaker-or-organizer/theodore-tso-google-0

2)
A binary is included in hex format and there is no copyright/license notice:
13M066-06/DRIVERS/MDIS_LL/M066/DRIVER/COM/m66_flex.h
13Z010-06/DRIVERS/MDIS_LL/QSPIM/DRIVER/COM/qspim_plddata.h

3)
MDISforLinux/INCLUDE/COM/MEN/mpc85xx.h
See also current version under https://github.com/torvalds/linux/blob/master/arch/powerpc/platforms/85xx/mpc85xx.h

I will give you feedback to this after examination.

dpfeuffer, 2019-08-21:
>There is no information about where this file has been derived from and the GPL version in its header is not specified:
13M077-90/DRIVERS/M077/DRIVER/serialP_m77.h

I found some similar files with same license text and suggest to stick to that license comment, because:
1) we are not allowed to change the license
2) it's too time consuming and no chance of success to contact the author for license clearing
see
https://raw.githubusercontent.com/spotify/linux/master/include/linux/serialP.h
https://git.kontron-electronics.de/linux/linux-imx-exceet/blob/de47725421ad5627a5c905f4e40bb844ebc06d29/include/linux/serialP.h

dpfeuffer, 2019-08-21:
>3.
>MDISforLinux/INCLUDE/COM/MEN/mpc85xx.h

This file is part of repo https://github.com/MEN-Mikro-Elektronik/mdis_include_com_men
I checked 13MD05-90 and 13MD05-60 repos that uses mdis_include_com_men as submodule. None of this master repos are using the mpc85xx.h header file. It is used from other MEN SW (MENMON).
So we can remove this header file from mdis_include_com_men repo.

@KonradJPE : Please remove mpc85xx.h from https://github.com/MEN-Mikro-Elektronik/mdis_include_com_men

dpfeuffer, 2019-08-21:
>2.
>A binary is included in hex format and there is no copyright/license notice:
13M066-06/DRIVERS/MDIS_LL/M066/DRIVER/COM/m66_flex.h
13Z010-06/DRIVERS/MDIS_LL/QSPIM/DRIVER/COM/qspim_plddata.h

We have to clarify in general how to deal with firmware binary blobs. This may take some time.

dpfeuffer, 2019-10-07:
>2
A binary is included in hex format and there is no copyright/license notice:
13M066-06/DRIVERS/MDIS_LL/M066/DRIVER/COM/m66_flex.h
13Z010-06/DRIVERS/MDIS_LL/QSPIM/DRIVER/COM/qspim_plddata.h
We have to clarify in general how to deal with firmware binary blobs. This may take some time.

License notes added with pull requests:
https://github.com/MEN-Mikro-Elektronik/13M066-06/pull/3
https://github.com/MEN-Mikro-Elektronik/13Z010-06/pull/1

closed 2019-11-18 by KonradJPE