MEN-Mikro-Elektronik/13MD05-90 #119: mdiswiz: system scan via GUI generates incomplete Makefile for G22

RBachmannMEN, 2019-05-13:
master@966a6ae

On the G22 board, using the `mdiswiz` GUI to scan the system sometimes (see below) results in an incorrect Makefile which does not include all necessary drivers.
Immediately after the scan from the GUI (`Build -> Scan`), the Makefile does include all necessary drivers. However, they are removed when saving the project.
The `scan_system.sh` script correctly generates the Makefile (however, the date in the Makefile header is wrong and seems to come from [Makefile.tpl](https://github.com/MEN-Mikro-Elektronik/13MD05-90/blob/master/MDISforLinux/BUILD/MDIS/TPL/DSC/Makefile.tpl)).

Steps to reproduce:

1. Open `mdiswiz` on a G22 board
2. Press the `Cancel` button in the `mdiswiz` popup
3. Run `Build -> Scan`
4. Open the generated Makefile, which includes the correct drivers
5. Save the `mdiswiz` project
6. Open the generated Makefile, which now does not include correct drivers

The incorrectly and correctly generated Makefiles are in this [github gist]:(https://gist.github.com/RBachmannMEN/9a3eee90d2f167954593ad105004e54e)

KonradJPE, 2019-05-15:
> The scan_system.sh script correctly generates the Makefile (however, the date in the Makefile header is wrong and seems to come from Makefile.tpl).

Header is fixed on jpe-dev branch. For further info please refer to issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/62

dpfeuffer, 2019-05-21:
The main problem is, that this lines which describe two native Linux drivers get lost when mdiswiz creates the Makefile:
```
ALL_NATIVE_DRIVERS = \
     DRIVERS/Z001_SMB/driver_g2x.mak\
     DRIVERS/CHAMELEON/driver.mak\
```
There is a 13z00190.xml file for the native Linux driver Z001_SMB (not a MDIS low-level driver!). This package description file lists only driver.mak but not the used driver_g2x.mak. The instance of this driver is also not visible within the mdiswiz device tree and the driver configuration is not written to system.dsc. I'm not aware how mdiswiz handles native Linux drivers in general. Please investigate the problem.

KonradJPE, 2019-06-19:
We have added this native drivers into G22 CPU section. 
Now when project with G22 is opened/created then this native drivers are added automatically by mdiswiz. As far as I know, this driver_g2x.mak is used only by G22, so there is no need to add this driver into xml file. Now it is hardcoded in scan_system, mdiswiz sources. 

@dpfeuffer  Is this solution acceptable?

dpfeuffer, 2019-06-24:
Yes, I think this is a pragmatical solution.

mlassota, 2019-06-28:
I've just update mdiswiz binary in 13MD05-90 to the latest version.

closed 2019-06-28 by mlassota