MEN-Mikro-Elektronik/13MD05-90 #121: mdiswiz doesn't consider Makefile created from scan_system.sh/mdiswiz

dpfeuffer, 2019-05-21:
We have actually the following mdiswiz/scan_system.sh related problems:
1. Software configuration settings made within mdiswiz are stored in the Makefile but not be considered, when the MDIS project is re-opened. See issue #106 
2. Makefiles generated from scan_system.sh are not considered from mdiswiz, when the MDIS project is re-opened. See issue #119 
3. We have redundant MEN board configuration parameters within mdiswiz and scan_system.sh. If new MEN boards shall be supported, both sources must be enhanced.    

The concept how mdiswiz and scan_system.sh works together is not well designed.
Suggestions for improvement:
A) Read back Makefile from mdiswiz (solves 1., may help to solve 2.)
B) Store software configuration within system.dsc file which is already read back. But this must be compatible to the used descgen tool, that creates the .bin files from system.dsc.  (solves 1.)
C) Complete rework of system scan functionality: System scan script shall gather all information about MEN HW at target system and stores this into an (json/yaml) config file. MDIS Wizard creates with the help of this config file system.dsc/Makefile. Which this approach, only the MDIS Wizard needs HW configuration parameters (solves 1..3).

What do you think about it?

KonradJPE, 2019-05-31:
After investigation I have to admit, that my statement 

> mdiswiz does not read Makefile at all, it is creating it from scratch every time.

was not entirely correct. 
mdiswiz is reading Makefile, but not all parameters. Because of that we will try to use solution 
A). 
We will try to  investigate why some groups of .mak files are missing (like NATIVE_DRIVER in issue https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/119). 
Maybe at some point it will be possible to get use our xml parser in mdiswiz.

Marcin is now updating our xml parser for scan_system script. Are there any other exceptions for Makefile like: 
- do not include internal tools as default,
- what about swmodules without `swap="false"` option?

We will appreciate your comments.

dpfeuffer, 2019-05-31:
>what about swmodules without swap="false" option?

From PackageDesc-Spec: *For Low Level Driver modules: If the LL driver has no swapped variant, the "swap" flag shall be set to "false".* 
If **swap="false"** is not present in the xml, then the driver has a swapped variant (driver*_sw.mak file). In this case, the MDIS wizard provides two driver variants (swapped and non-swapped) and the proper driver variant must be built. This depends on the CPU-->Device path (if there is a swapping bridge between the CPU and the device). I think this info is board specific available within the MDIS wizard. 

Currently I'm not aware of other exceptions.

KonradJPE, 2019-07-31:
This issue is connected with: #126. 
When issue #126 is closed, then differences between Makefile, system.dsc generated by scan_system.sh and mdiswiz should be irrelevant. Then futher tests regarding this issue will be performed.

KonradJPE, 2019-08-08:
We have found, that PCI/DRIVER/COM/driver.mak driver is included in Makefile.tpl. But it is not added by mdiswiz as default. 

So question is: 
Should we remove or leave this driver from default build (scan_system.sh)?

dpfeuffer, 2019-08-09:
Please keep it until now, because it may have a reason to be there. Maybe we should add the PCI BBIS driver to mdiswiz.
Please create a new issue (addressed to me) for the clarification when you close this issue.

dpfeuffer, 2019-08-09:
Today, in our telephone conference, I had the following thought in my mind when you told us about your plan of restructuring scan_system.sh: 
> C) Complete rework of system scan functionality: System scan script shall gather all information about MEN HW at target system and stores this into an (json/yaml) config file. MDIS Wizard creates [...]
But this suggestion for improvement (see the first comment in this issue) is lot of work.

KonradJPE, 2019-08-09:
After closing issue #126 it seems that such a big interfere into scan_system script is unnecessary. 
Now scan_system.sh script works pretty similar to mdiswiz. 
system.dsc and Makefies are consistent with each other, no matter if it was created by mdiswiz or scan_system. 
Data from .xml files are obtained during scanning - except from below xml-s:
13y00206.xml - F14_MON
13xm0106.xml - XM01BC
13ad7806.xml - AD78
13y00406.xml - SMB2
13xc0206.xml - XC02
13p7c906.xml - PI7C9_GPIO
13y00106.xml - LM63
13pp0406.xml - PP04
13y00706.xml - F14BC

dpfeuffer, 2019-08-09:
>After closing issue #126 it seems that such a big interfere into scan_system script is unnecessary.

I agree with you.

>Data from .xml files are obtained during scanning - except from below xml-s:
13y00206.xml - F14_MON
...

What is the reason that these package desc files are not scanned?
Is the content of these files still hardcoded in scan_system?

mlassota, 2019-08-09:
> What is the reason that these package desc files are not scanned?

First modification regarding XML parsing was about extracting M-Module data (all 13m*.xml files). Second was about extracting IP core data (all 13z*.xml files).
The rest of files do not contain data related to M-Modules and IP cores. Their exclusion speeds up the scanning process.

> Is the content of these files still hardcoded in scan_system?
Some of them are hardcoded, other are not.

KonradJPE, 2019-08-09:
We are unsure when it comes to below devices (how to detect, and add valid .mak or descriptors):
13y00206.xml - F14_MON
13ad7806.xml - AD78
13xc0206.xml - XC02
13y00106.xml - LM63
The rest is hardcoded and prescribed to certain CPU devices/boards, and we will have to think if there is possibility to detect them automatically.

dpfeuffer, 2019-08-09:
Ok, thanks for your explanation, that seems pragmatic and ok to me.

dpfeuffer, 2019-08-09:
>We are unsure when it comes to below devices (how to detect, and add valid .mak or descriptors):
13y00206.xml - F14_MON
13ad7806.xml - AD78
13xc0206.xml - XC02
13y00106.xml - LM63

13y00206.xml - F14_MON
The f14_mon driver uses the PCI BBIS.
Back to your question from this morning:
>We have found, that PCI/DRIVER/COM/driver.mak driver is included in Makefile.tpl. But it is not added by mdiswiz as default.
So question is:
Should we remove or leave this driver from default build (scan_system.sh)?

The PCI BBIS driver shall only be added from scan_system.sh if the ```<busif>PCI</busif>``` is used (see 13y00206.xml). I assume that is the functionality that mdiswiz has implemented.

The other driver packages contains drivers for SMB devices.
For Windows I implemented a detection mechanism into an Installshield script for these SMB devices. Each SMB device has a fix SMB device address, and the script probes each of these addresses with a SmbReadByte from offset 0x00, to see if there is any device or not. 
This little tool is used https://github.com/MEN-Mikro-Elektronik/13Y004-06/blob/master/DRIVERS/MDIS_LL/SMB2/TOOLS/SMB2_TOUCH/COM/smb2_touch.c

mlassota, 2019-08-09:
We have no hardware to test it, so I think we leave it as it is.

dpfeuffer, 2019-08-09:
>We have no hardware to test it, so I think we leave it as it is.

Ok, for the SMB devices.

But for the PCI BBIS: Please add these driver from scan_system.sh only if it is required (if the ```<busif>PCI</busif>``` is used). I did a grep and found only a match for 13y00206.xml.

mlassota, 2019-10-01:
> But for the PCI BBIS: Please add these driver from scan_system.sh only if it is required (if the <busif>PCI</busif> is used). I did a grep and found only a match for 13y00206.xml.

Still, we have no hardware to test it. According to 13y00206.xml this monitoring hardware can be found on F14/F15/F17/F18/D601/SC14 only.

dpfeuffer, 2019-10-02:
> We have found, that PCI/DRIVER/COM/driver.mak driver is included in Makefile.tpl. But it is not added by mdiswiz as default.
So question is:
Should we remove or leave this driver from default build (scan_system.sh)?

Because F14/F15/F17/F18/D601/SC14 is EOL/EOS:
Please remove PCI/DRIVER/COM/driver.mak from ALL_BB_DRIVERS in Makefile.tpl. This will prevent to build unused SW and speed up the build process a little bit. If the PCI BBIS is required from a customer for an old project, it is still possible to add it from the MDIS Wizard.

mlassota, 2019-10-02:
> Please remove PCI/DRIVER/COM/driver.mak from ALL_BB_DRIVERS in Makefile.tpl

Done.

closed 2019-10-02 by mlassota