MEN-Mikro-Elektronik/13MD05-90 #123: 15P506 CANCLOCK issue

bmoore84, 2019-06-26:
When using a P506, mdiswizard generates a system.dsc where "CANCLOCK" is set with the wrong value. There is no option to select a P506. Customer must manually change the CANCLOCK as a workaround.

In the older versions of MDIS, the P506 was an option that you could choose in mdiswiz. This has since been removed and now when you select the MEZZ_CHAM option, and then add the CAN driver, it uses a template that only works for the F215/G215.
An option to select the P506 in mdiswiz should be added to avoid this issue,

dpfeuffer, 2019-06-27:
The various Chameleon Carrier models like P506 were replaced by one generic MEZZ_CHAM model at 2014. The reason was to reduce the maintaining costs of the MDIS wizard.
The MEZZ_CHAM model works for most MEN boards with Chameleon FPGAs without special settings - not only for F215/G215.
If it is necessary to set a special "CANCLOCK" for the P506, we should note it in the user documentation (P506 HW manual?). I know it would be more comfortable for customers to have a dedicated P506 model, but it must be implemented, tested, maintained.
@Florian: What is your opinion?

FlorianRauh, 2019-06-27:
I guess it´s not worth the effort doing it in mdiswiz.
@Blake as discussed please add all needed information in the HW UserManual
Issue shall be closed.

closed 2019-07-05 by KonradJPE