MEN-Mikro-Elektronik/13MD05-90 #124: mt_shmem and mt_bench tools from mdis_ll_mt don't compile on Linux

mlassota, 2019-07-01:
Are these tools meant to be compiled on Linux? But I see no exclusuion in XML file.

* For mt_shmem compilation it lacks definition of UOS_SHMEM_HANDLE type. Seems like external library missing since there's no reference in 13MD05-90 for it.

* For mt_bench compilation it lacks definition of MAX_OPENS, CALLS and OPENS constants. These are defined for all supported OSes but Linux.

dpfeuffer, 2019-07-02:
**mt_shmem**
MDIS for Linux/Windows supports no shared mem functionality (usr_oss.h defines NO_SHARED_MEM). Please add notos tags to mt_shmem in mt.xml (like mt_callback, see #68).
```
    <notos>Linux</notos>        
    <notos>Windows</notos> 
```

**mt_bench**
Add
```
#ifdef LINUX
# define MAX_OPENS 1000	/* max number of opened paths */
# define CALLS   10000  /* number of function callcnt */
# define OPENS   10     /* number of opened paths */
#endif LINUX
```
and test if mt_bench succeeds with these values.

mlassota, 2019-07-02:
> mt_shmem

Added. No problems.

> mt_bench

Added. Compiles with no problems.

closed 2019-07-02 by mlassota