MEN-Mikro-Elektronik/13MD05-90 #125: sja1000_test doesn't compile

mlassota, 2019-07-01:
Constant UOS_SIG_MAX is defined nowhere that leads to the following errors:
```
13MD05-90_Install/DRIVERS/MDIS_LL/SJA1000/TEST/SJA1000_TEST/COM/sja1000_test.c:79:4: error: #error "*** UOS_SIG_USR3 not supported for this MDIS implementation!"
 #  error "*** UOS_SIG_USR3 not supported for this MDIS implementation!"
    ^~~~~
13MD05-90_Install/DRIVERS/MDIS_LL/SJA1000/TEST/SJA1000_TEST/COM/sja1000_test.c:86:4: error: #error "*** UOS_SIG_USR4 not supported for this MDIS implementation!"
 #  error "*** UOS_SIG_USR4 not supported for this MDIS implementation!"
    ^~~~~
```
Is this a non-Linux tool or we're missing something here?

dpfeuffer, 2019-07-02:
In the past, we had an OS specific UOS_SIG_MAX define as well as UOS_SIG_USR3/4 in INCLUDE/NATIVE/MEN/usr_os.h. This was skipped because of Linux limitation to only two available user signals.
Please change the code from
```
#ifndef UOS_SIG_USR3
#	define UOS_SIG_USR3 (UOS_SIG_USR2 + 1)
#	if (UOS_SIG_USR3 > UOS_SIG_MAX)
#		error "*** UOS_SIG_USR3 not supported for this MDIS implementation!"
#	endif	
#endif
#ifndef UOS_SIG_USR4
#	define UOS_SIG_USR4 (UOS_SIG_USR3 + 1)
#	if (UOS_SIG_USR4 > UOS_SIG_MAX)
#		error "*** UOS_SIG_USR4 not supported for this MDIS implementation!"
#	endif	
#endif
```
to
```
#ifndef UOS_SIG_USR3
#	define UOS_SIG_USR3 (UOS_SIG_USR2 + 1)
#	define WARN_UOS_SIG_USR 1
#endif
#ifndef UOS_SIG_USR4
#	define UOS_SIG_USR4 (UOS_SIG_USR3 + 1)
#	define WARN_UOS_SIG_USR 1
#endif
```
and add
```
#ifdef WARN_UOS_SIG_USR
	printf("*** WARNING: Signal handling UOS_SIG_USR3/4 may not work correct due to OS limitation\n");
#endif
    /*--------------------+
    |  check arguments    |
    +--------------------*/
```

mlassota, 2019-07-02:
Changes made. Compiles with no problems.

closed 2019-07-02 by mlassota