MEN-Mikro-Elektronik/13MD05-90 #126: Trivial scanning of driver makefiles in XMLs for IPCores in scan_system.sh

mlassota, 2019-07-04:
There's this trivial part
~~~~
lldrv=`fgrep ".mak" $xmlfile | grep DRIVER | sed "s/<makefilepath>//;s/<\/makefilepath>//"`
G_makefileLlDriver+=" "$lldrv
~~~~
in scan_system.sh. It causes the final Makefile to contain all the low-level drivers from particular XML file instead of those only related to particular IPCore (device id). Opening then and saving project in MDIS Wizard removes these extra drivers from Makefile making the difference between how Scanner and Wizard work.
Maybe we could do the similar to https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/85 and use DevId from Chameleon Table with the chamv2id tag from XML to get only device related drivers?

dpfeuffer, 2019-07-05:
Thanks for your hint.
I think only the driver that will be specified in system.dsc should be added to the Makefile. The other drivers will not be used because they are not in system.dsc. Please fix it.
Now, I see another question/problem. For example, in 13z01706.xml we have three driver variants that match to ```<chamv2id>127</chamv2id>```. Which driver variant will be added to system.dsc from scan_system.sh? Just the first one? During manual MDIS system configuration with mdiswiz, the user have to determine the driver version that he want to use.

mlassota, 2019-07-05:
> For example, in 13z01706.xml we have three driver variants that match to <chamv2id>127</chamv2id>. Which driver variant will be added to system.dsc from scan_system.sh?

Well, it seems that 13Z127 is not detected by scan_system.sh at all. That's because there are IP Core names hardcoded in scan_system.sh except 13Z127 and there's no template for it too. Maybe there are other missigng, I didn't check. It needs to be fixed.
But back to your question. In this case a user has a free will to choose what driver he wants to build (correct me if I'm wrong) and we have to decide for him, so either the first driver or all drivers is a good choice - it's up to you to decide.
As for the <chamv2id>34</chamv2id> there are two drivers: memory mapped io and io mapped io. Correct me if I'm wrong but I think that Chameleon table has all the information needed to choose the correct driver.
An excerpt from Chameleon table:
~~~~
BARs of FPGA PCI device 04:14.0:
  BAR0: 0xb8b00000; size: 0x00000000, mapType: MEM;
  BAR1: 0x00005001; size: 0x00000000, mapType: IO;
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
  5 0x0022 16Z034_GPIO                0    0   0  10  63   0 0x00000180 0xb8b00180
~~~~
By substracting Offset from Address of 16Z034_GPIO we get BAR address and its corresponding mapType, MEM in that case, so the default driver should be used.

dpfeuffer, 2019-07-08:
Because you are already parsing the driver package description files for M-Module matching, why don't use the same mechanism for IP core matching? Could you please evaluate the possibility and effort to change from hard coded IP core names to a dynamic collection.
For clarification: 13* indicates driver software and 16* IP cores. There is no 13Z127 driver package, but an 16Z127 IP core. I assume this was just a typo.
To add all matching drivers in Makefile doesn't make sense, because only one driver could be assigned for each IP core in system.dsc. To add just the first matching driver could be a possible solution. Another suggestion: How about asking the user during the scan to choose one of all matching drivers? Please evaluate the possibility and effort for this.
To decide between z17 and z17_io driver, the mapType MEM/IO could be used. If we change from hard coded to dynamic enumeration, you could parse all driver names for "_io". Then a xxx_io driver could be assigned if mapType=IO, otherwise the xxx driver.

mlassota, 2019-07-08:
> For clarification: 13* indicates driver software and 16* IP cores. There is no 13Z127 driver package, but an 16Z127 IP core. I assume this was just a typo.

Sorry, my bad.

> Another suggestion: How about asking the user during the scan to choose one of all matching drivers?

What about MDIS Wizard then? Currently it does not support user input while scanning the system? Would you like it to be interactive too?

dpfeuffer, 2019-07-08:
> What about MDIS Wizard then? Currently it does not support user input while scanning the system. Would you like it to be interactive too?
Good remark. Maybe this wouldn't be possible at all. Do you know?

mlassota, 2019-07-09:
>> What about MDIS Wizard then? Currently it does not support user input while scanning the system. Would you like it to be interactive too?

> Good remark. Maybe this wouldn't be possible at all. Do you know?

That should be possible. I would expect the behavior similar to that in terminal when running script i.e. editing is not possible but you can enter data.
However currently in MDIS Wizard you can't select IO driver, which means that when you open such a system.dsc having IO driver it would revert it to the default driver.
So maybe we stick to the option with first driver only. How do you think?

dpfeuffer, 2019-07-11:
>That should be possible. I would expect the behavior similar to that in terminal when running script i.e. editing is not possible but you can enter data.

Good,

>However currently in MDIS Wizard you can't select IO driver, which means that when you open such a system.dsc having IO driver it would revert it to the default driver.
So maybe we stick to the option with first driver only. How do you think?

Except for a few customer specific cases, all Chameleon Units in FPGAs@PCI(e) are memory mapped. For the customer specific cases, we have to provide a manual created MDIS configuration to the customer and I would not support these cases within MDIS Wizard or scan_system.sh.
For FPGAs@ISA, there is a ModBbisChamIsa class, that considers if _io driver variants are required (as I understand it right). 
But independent of the mapping type, in some cases there are more than one driver match for a HW device. To stick to the 13z01706.xml example, we have three driver variants that match to the 16Z127 IP core:
```
		<model>
			<hwname>16Z127_GPIO</hwname>
			<description>32bit GPIO controller in Chameleon FPGA</description>
			<autoid>
				<chamv2id>127</chamv2id>
			</autoid>
			<swmodulelist>
				<swmodule>
					<name>z17_z127</name>
					<description>GPIO Driver for Z127_GPIO</description>
					<makefilepath>Z017/DRIVER/COM/driver_z127.mak</makefilepath>
				</swmodule>
			</swmodulelist>
		</model>
		<model>
			<hwname>16Z127_GPIO</hwname>
			<modelname>16Z127_GPIO_NOIRQ</modelname>
			<description>32bit GPIO controller in Chameleon FPGA (without IRQs)</description>
			<autoid>
				<chamv2id>127</chamv2id>
			</autoid>
			<swmodulelist>
				<swmodule>
					<name>z17_z127_noirq</name>
					<description>GPIO Driver for Z127_GPIO without IRQs</description>
					<makefilepath>Z017/DRIVER/COM/driver_z127_noirq.mak</makefilepath>
				</swmodule>
			</swmodulelist>
		</model>
		<model>
			<hwname>16Z127_GPIO</hwname>
			<modelname>16Z127V01_GPIO</modelname>
			<description>32bit GPIO controller (16Z127-01) in Chameleon FPGA (with Variable Debounce Time)</description>
			<autoid>
				<chamv2id>127</chamv2id>
			</autoid>
			<swmodulelist>
				<swmodule>
					<name>z17_z127v01</name>
					<description>GPIO Driver for Z127V01_GPIO with Variable Debounce Time </description>
					<makefilepath>Z017/DRIVER/COM/driver_z127v01.mak</makefilepath>
				</swmodule>
			</swmodulelist>
		</model>
```

I propose to use the standard driver variant (model without ```<modelname>xxx</modelname>``` tag) as default, but to provide the possibility to choose an other matching driver variants by asking the user from scan_system.sh if an optional driver variant should be used. To make the decision easier, the ```<swmodulelist>/<name> + <description>``` should be shown for the selection.

FlorianRauh, 2019-07-24:
1) 

> I propose to use the standard driver variant (model without <modelname>xxx</modelname> tag) as default, but to provide the possibility to choose an other matching driver variants by asking the user [...]

I fully agree to Dieter. Marcin and Konrad pls try to do it like that.

2) 

> However currently in MDIS Wizard you can't select IO driver, 

In my opinion , thats not true as I´m able to choose MEM or IO mapped driver variants in mdiwiz

[image]

MEM / IO mapped driver variants: shall be treated like 1)

mlassota, 2019-08-08:
Done

closed 2019-08-08 by mlassota