MEN-Mikro-Elektronik/13MD05-90 #130: LIBSRC subfolders: inconsistent structure and documentation files missing

dpfeuffer, 2019-07-05:
In 13MD05-90/MDISforLinux/LIBSRC, some libs are added as submodules some not.
Documentation files missing, example OSS:
- COM/oss_specification.c missing
- DOC/Doxyfile missing
- DOC/html/ outdated

KonradJPE, 2019-07-05:
> In 13MD05-90/MDISforLinux/LIBSRC, some libs are added as submodules some not.

Should we move each LIBSRC/dir into separate submodule?

dpfeuffer, 2019-07-05:
Currently not, I have to think about it later and to provide you the missing documentation files. Tbd.

dpfeuffer, 2019-10-10:
The following libs are currently directly located in the 13MD05-90 repo (not as submodule):
A) Tightly bound to MDIS, location (ok):
   - BBIS_KERNEL
   - MDIS-API
   - MDIS_COMPONENT_COMMON
   - MDIS_KERNEL

B) Not directly MDIS specific, --> **please moved into submodules**:
   - DBG (dbgu.c contains Windows specific parts and includes MDIS headers. I don't know why, but I would let is untouched to be on the safe side.) 
   - OSS
   - UTI
   - VME4L_API

See also MEN_13MD05-90_SA_0275 [DES] Submodule Repo Usage

I changed this issue title and will create a new issue about documentation.

dpfeuffer, 2019-10-18:
The following repos are Linux specific. Therefore, we have to append the prefix "_linux" to be able to store other OS implementations of this repos:
- mdis_libsrc_dbg -> mdis_libsrc_dbg_linux
- mdis_libsrc_usr_oss -> mdis_libsrc_usr_oss_linux **: renaming not yet possible!**
  This repo isn't new. When we rename it, we would run into trouble if someone tries to checkout any older commit of 13MD05-90. So we have to leave it as it is until the GitHub 2 GitLab migration.
- mdis_libsrc_oss -> mdis_libsrc_oss_linux
- mdis_libsrc_vme4l_api : is ok, because the term "wme4l"

Sorry that I forget to mention it!
Please add also to the README.md files "XXX library for Linux"

KonradJPE, 2019-10-21:
Renamed: 
mdis_libsrc_dbg -> mdis_libsrc_dbg_linux
mdis_libsrc_oss -> mdis_libsrc_oss_linux

> Please add also to the README.md files "XXX library for Linux"

Added to:
mdis_libsrc_dbg_linux
mdis_libsrc_oss_linux
mdis_libsrc_usr_oss

dpfeuffer, 2019-10-21:
Thanks for the fix.

closed 2019-10-21 by dpfeuffer